chipsalliance/chisel #582: Provide API for Casting Bundle to Parent Type for Bulk Connect

jackkoenig, 2017-04-18:
I thought there was an issue for this already but I could find it. This is useful when you have some Bundle Type that inherits from another type to add some fields but wants to be bulk connected somewhere as if it is the parent type.

mwachs5, 2017-04-18:
👍

colinschmidt, 2017-04-18:
This would be great! The current workarounds I've been using are not very nice...

jackkoenig, 2019-10-07:
Context for anyone wanting to tackle this issue.

You can fairly easily implement Chisel2-style bulk connect using `Record.elements` (returns `ListMap[String, Data])` and `DataMirror.directionOf` (to get directions).

The tricky part is trying to do something "safer". There's definitely utility in allowing a Bundle to act like a super type (Liskov substitution principle), thus allowing some type of safe casting is an alternative, more complicated solution. I don't know how one might do this non-reflectively, so it could get pretty messy.

jackkoenig, 2022-03-22:
Fixed in 3.5.0 via DataView (`.viewAsSupertype`)

closed 2022-03-22 by jackkoenig