chipsalliance/chisel #584: Graphical testbench for Chisel

chiselmonster, 2017-04-20:
Hi, just wondering if there is such a tool which allows me to use the peek function to generate some time waveform based on the clock cycles. 

Currently trying to develop some Chisel code but finding it quite difficult to use the existing iotesters to verify the results

edwardcwang, 2017-04-20:
Yes, the PeekPoke tester will dump a VCD file which you can open in a graphical viewer.

closed 2017-04-20 by edwardcwang

chiselmonster, 2017-04-20:
Hi Edward,
I am new to Chisel, may I know if I were to use the default simulator, does it dump out the vcd file? I had some trouble setting up verilator on windows

chick, 2017-04-20:
I believe it is off by default.  See: src/test/scala/examples/GCDSpec.scala
<https://github.com/ucb-bar/chisel-testers/blob/master/src/test/scala/examples/GCDSpec.scala>
for
an example of how to turn it on.
-chick

Chick Markley
Staff Programmer -- Aspire Lab
University of California, Berkeley
583-2 Soda Hall, Berkeley, CA 94720-1776
chick@berkeley.edu

edwardcwang, 2017-04-20:
Yes, the firrtl backend seems to have vcd off by default, though it seems to be on by default for verilator.