chipsalliance/chisel #585: Latency delay in VCD file

chiselmonster, 2017-04-21:
Hi,

I did a simulation with the following code

```
val count_hold_delay_reg = RegInit(UInt(8.W), 0.U())
  
when(div_sig === 1.U) {
    count_hold_delay_reg := ((pwm_in_high_reg*256.U)/ pwm_in_period_reg)
  }
io.pwm_in_count_out := count_hold_delay_reg
```
After simulation and dumping out of the VCD, I see one clock latency in the io.pwm_in_count_out signal

[image]

Both nodes are essentially the same node in Verilog, how come I see the one clock latency whenever I assign something to the IO?

Thanks

shunshou, 2017-04-21:
Out of curiosity, are you using Firrtl interpreter or Verilator to run tests? I had this exact same problem ~3 years ago with an older version of Chisel... don't believe I see this problem in the version of chisel3 I'm using... at least with Verilator... so it could be some bad interpretation on the Firrtl interpreter part if you're using that for simulation (same with the Chisel2 problem -- someone decided to simulate w/ behavior that didn't make sense from the POV of someone writing hardware)...

chiselmonster, 2017-04-21:
Oh you are right on, I am using Firrtl on Chisel3
I had some problem installing Verilator as I am using Win system.

chick, 2017-05-02:
Just to keep this current, I have found the problem associated with the issue raised here but that discovery has revealed a couple of other issues that I am tracking down now.  I expect to have a PR for this in a couple of days

chiselmonster, 2017-05-03:
Noted chick! Thanks for the updates, currently I migrated to a Linux system to run my verilator, no issue there so far!

closed 2017-06-19 by azidar