chipsalliance/chisel #586: issue on probing different nodes for the same signal give different waveform

edysusanto, 2017-04-21:
Hi,

I coded a simple counter and probe the signal at the io boundary and at the flipflop itself.
It shows two different waveforms on GTK.

May I know if there is somebody can point out what is wrong with my code?

```
class pwminCount extends Module {
  val io = IO(new Bundle {
    // Outputs
    val testReg = Output(UInt(4.W))
    // Inputs
  })

  // For testing only
  val testReg = RegInit(UInt(4.W), 0.U)
  testReg := testReg + 1.U
  io.testReg := testReg

}

class pwminCountTests(c: pwminCount) extends PeekPokeTester(c) {
  println("\ntesting pwminCount...\n")
  var finishTime = 50

    for(i <- 0 until finishTime){
      println("test=" + peek(c.io.testReg))
    step(1)
  }
}
```

[image]

chick, 2017-04-25:
I will take a look at this in the next day or so, I suspect this is an artifact of the interpreters synthetic view of timing.  Can you confirm, that this is occurs when using the interpreter back end

edysusanto, 2017-04-26:
Thanks for your help! Yes, I use Firrtl to simulate and use GTK to view the waveform.

Here is the code that showing the Firrtl part.
```
object pwminCount {
  def main(args: Array[String]): Unit = {
    val manager = new TesterOptionsManager {
      testerOptions = testerOptions.copy(backendName = "firrtl", testerSeed = 7L)
      interpreterOptions = interpreterOptions.copy(setVerbose = false, writeVCD = true)
    }
    iotesters.Driver.execute(() => new pwminCount, manager) {
      c => new pwminCountTests(c)
    }
  }
}
```

azidar, 2018-03-16:
We believe this is fixed. Please reopen if issue was not resolved.

closed 2018-03-16 by azidar