chipsalliance/chisel #588: Recursively SetFirrtlDirection for Aggregate types

jchang0, 2017-04-21:
Hi,
  I would like to set the direction to override any previously direction set in elements of a bundle.

The use is to create a monitor bundle which will turn all signals into inputs so I can create a monitor module that only observe the bits on a bundle, but not drive it.

Here is a snippet sample code it currently generates from a DecoupledIO if I specify Input(out_bundle)
    output io : {flip out_bundle : {aw : {flip ready : UInt<1>, valid : UInt<1>, bits : {id : UInt<8>}}}

Currently, there are some fields that are flipped twice, so out_bundle.ready is it's actually output rather than the input that I wanted. (this causes CheckGender error errors)

I see that the direction is recursively set for Vecs:

(in Data.scala)
  private[chisel3] def setFirrtlDirection(target: Data, direction: Direction): Unit = target match {
    case (vec: Vec[Data @unchecked]) => vec.sample_element.firrtlDirection = direction
    case _ => target.firrtlDirection = direction
  }

Can we do the same for other aggregate data types? Or if there is a good reason not to recursively set the fields, create another API (e.g. InputRecursive(out_bundle) ) that will force all the elements to the same direction?

Thanks,
Jacob

jchang0, 2017-04-21:
Ah, I just saw #578 looks like it's related to it.

jchang0, 2017-04-27:
Just for discussion: I added the following to Data.scala in my local chisel3 and it seems to do what I wanted:  (I created a new type called Monitor (name may be overloaded, might have to use another name)

diff --git a/chiselFrontend/src/main/scala/chisel3/core/Data.scala b/chiselFrontend/src/main/scala/chisel3/core/Data.scala
index c17672f..1745632 100644
--- a/chiselFrontend/src/main/scala/chisel3/core/Data.scala
+++ b/chiselFrontend/src/main/scala/chisel3/core/Data.scala
@@ -108,6 +108,15 @@ object Flipped {
   }
 }
 
+object Monitor {
+  def apply[T<:Data](source: T)(implicit compileOptions: CompileOptions): T = {
+    val target = source.chiselCloneType
+    Data.clearFirrtlDirection(target)
+    Data.setFirrtlDirection(target, Direction.Input)
+    Binding.bind(target, InputBinder, "Error: Cannot set as input ")
+  }
+}
+
 object Data {
   /**
   * This function returns true if the FIRRTL type of this Data should be flipped
@@ -116,7 +125,7 @@ object Data {
   * Note that the current scheme only applies Flip to Elements or Vec chains of
   * Elements.
   *
-  * A Record is never marked flip, instead preferring its root fields to be marked
+  * A Record is never marked fli  d p, instead preferring its root fields to be marked
   *
   * The Vec check is due to the fact that flip must be factored out of the vec, ie:
   * must have flip field: Vec(UInt) instead of field: Vec(flip UInt)
@@ -153,6 +162,14 @@ object Data {
     case _ => target.firrtlDirection = direction
   }
 
+  private[chisel3] def clearFirrtlDirection(target: Data): Unit = {
+    target.firrtlDirection = Direction.Unspecified
+    target match {
+      case (agg: Aggregate) => { agg.getElements.foreach( (elem: Data) => {Data.clearFirrtlDirection(elem)}) }
+      case _ => {}
+    }
+  }
+
   implicit class AddDirectionToData[T<:Data](val target: T) extends AnyVal {
     def asInput(implicit opts: CompileOptions): T = {
       if (opts.deprecateOldDirectionMethods)

jackkoenig, 2017-04-27:
As mentioned in the discussion of #440 (but not addressed with its own issue), `Input(...)` and `Output(...)` should both coerce the directions of a bundle. However, there is currently a bug in a mismatch between internal firrtlDirection and the direction binding. #578 is a discussion of implementation of solution and we should probably up the priority of this since some people may be taking advantage of incorrect "functionality"

I'm not sure about connecting from a bidirectional bundle to a single-directioned or undirectioned version of the same but there probably should be a way to do this with `:=` maybe?

ducky64, 2017-04-28:
I thought the current behavior was that while `Input(...)` and `Output(...)` emitted insane FIRRTL, the FIRRTL compiler does error out (so the big directionality refactor shouldn't break any code)?

jackkoenig, 2017-04-28:
It's wrong, but I don't think it's insane. I think it's possible to create cases where it works, it's just likely to error in Chisel

jchang0, 2017-04-28:
Before, I had thought Input implies the direction of the "data" portion of the bundle even if some signals (ready) might be an output out of that bundle, and have written code to that effect.  The reason why I through so was because it actually works correctly (even in simulation) when I interpret the signals as such (and does not work correctly if I interpret Input(Bundle) meant all signals are inputs, including the ready signals).

azidar, 2017-07-24:
Fixed in #617.

closed 2017-07-24 by azidar