chipsalliance/chisel #591: how to run Chisels sims using Verilator

edysusanto, 2017-04-25:
Hi, I am trying to run my Chisels code using Verilator, instead of Firrtl. However, I encountered error. 
Anyone can help me to solve this?

This is my tester code:
```
    val args = Array("--backend-name", "verilator")
    iotesters.Driver.execute(args, () => new pwminCount) { c =>
      new pwminCountTests(c)
    }
```

And this is the error that I got.

```
[info] Running pwminCount.pwminCount
[info] [0.002] Elaborating design...
[info] [0.045] Done elaborating.
verilator --cc pwminCount.v --assert -Wno-fatal -Wno-WIDTH -Wno-STMTDLY --trace
-O1 --top-module pwminCount +define+TOP_TYPE=VpwminCount +define+PRINTF_COND=!pw
minCount.reset +define+STOP_COND=!pwminCount.reset -CFLAGS -Wno-undefined-bool-c
onversion -O1 -DTOP_TYPE=VpwminCount -DVL_USER_FINISH -include VpwminCount.h -Md
ir C:\Users\edy\IdeaProjects\chisel-pwminCount\test_run_dir\pwminCount.pwminCoun
t237465786 --exe C:\Users\edy\IdeaProjects\chisel-pwminCount\test_run_dir\pwminC
ount.pwminCount237465786\pwminCount-harness.cpp
[error] (run-main-0) java.io.IOException: Cannot run program "verilator": Create
Process error=2, The system cannot find the file specified
java.io.IOException: Cannot run program "verilator": CreateProcess error=2, The
system cannot find the file specified
        at java.lang.ProcessBuilder.start(Unknown Source)
        at scala.sys.process.ProcessBuilderImpl$Simple.run(ProcessBuilderImpl.sc
ala:69)
        at scala.sys.process.ProcessBuilderImpl$AbstractBuilder.run(ProcessBuild
erImpl.scala:98)
        at scala.sys.process.ProcessBuilderImpl$AbstractBuilder.$bang(ProcessBui
lderImpl.scala:112)
        at chisel3.iotesters.setupVerilatorBackend$.apply(VerilatorBackend.scala
:252)
        at chisel3.iotesters.Driver$$anonfun$execute$1.apply$mcZ$sp(Driver.scala
:50)
        at chisel3.iotesters.Driver$$anonfun$execute$1.apply(Driver.scala:35)
        at chisel3.iotesters.Driver$$anonfun$execute$1.apply(Driver.scala:35)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
        at chisel3.iotesters.Driver$.execute(Driver.scala:35)
        at chisel3.iotesters.Driver$.execute(Driver.scala:89)
        at pwminCount.pwminCount$.main(pwminCount.scala:47)
        at pwminCount.pwminCount.main(pwminCount.scala)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.lang.reflect.Method.invoke(Unknown Source)
Caused by: java.io.IOException: CreateProcess error=2, The system cannot find th
e file specified
        at java.lang.ProcessImpl.create(Native Method)
        at java.lang.ProcessImpl.<init>(Unknown Source)
        at java.lang.ProcessImpl.start(Unknown Source)
        at java.lang.ProcessBuilder.start(Unknown Source)
        at scala.sys.process.ProcessBuilderImpl$Simple.run(ProcessBuilderImpl.sc
ala:69)
        at scala.sys.process.ProcessBuilderImpl$AbstractBuilder.run(ProcessBuild
erImpl.scala:98)
        at scala.sys.process.ProcessBuilderImpl$AbstractBuilder.$bang(ProcessBui
lderImpl.scala:112)
        at chisel3.iotesters.setupVerilatorBackend$.apply(VerilatorBackend.scala
:252)
        at chisel3.iotesters.Driver$$anonfun$execute$1.apply$mcZ$sp(Driver.scala
:50)
        at chisel3.iotesters.Driver$$anonfun$execute$1.apply(Driver.scala:35)
        at chisel3.iotesters.Driver$$anonfun$execute$1.apply(Driver.scala:35)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
        at chisel3.iotesters.Driver$.execute(Driver.scala:35)
        at chisel3.iotesters.Driver$.execute(Driver.scala:89)
        at pwminCount.pwminCount$.main(pwminCount.scala:47)
        at pwminCount.pwminCount.main(pwminCount.scala)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.lang.reflect.Method.invoke(Unknown Source)
[trace] Stack trace suppressed: run last compile:run for the full output.
java.lang.RuntimeException: Nonzero exit code: 1
        at scala.sys.package$.error(package.scala:27)
[trace] Stack trace suppressed: run last compile:run for the full output.
[error] (compile:run) Nonzero exit code: 1
[error] Total time: 4 s, completed 25 Apr, 2017 4:19:12 PM
```

jackkoenig, 2017-04-25:
Do you have Verilator installed? If not, check out the installation instructions (just following the ones for Verilator of course): https://github.com/ucb-bar/chisel3#installation

(Note, on Ubuntu-like machines, the version of Verilator install in apt-get will **not** work. You must install from source)

edysusanto, 2017-04-27:
hi jackkoenig, yes, I have Verilator and IntelliJ installed on windows. However, it seems my IntelliJ cannot see the Verilator.

ducky64, 2017-11-08:
This seems like a usage question not related to Chisel engineering. We'd appreciate that support questions are put on StackOverflow so they're searchable by others who run into the same issue.

If there's a Chisel code change you'd like to see, feel free to re-open.

closed 2017-11-08 by ducky64

zhangshengshan, 2023-04-11:
@jackkoenig  The installation of verilator must be from source code on Ubuntu-like systems , i think this tip should be in the doc or readme file , it can save a lot of time for env establishment