chipsalliance/chisel #593: Need CompileOptions in chisel3 utils

ucbjrl, 2017-04-25:
Now that we no longer assume NotStrict CompileOptions in chisel internals, more functions and utilities will need access to the client's CompileOptions. Failure to do so will provoke errors such as:
```
[error] .../hardfloat/src/main/scala/MulAddRecFN.scala:205: too many arguments for method do_zext: (implicit sourceInfo: chisel3.internal.sourceinfo.SourceInfo)chisel3.core.SInt
[error]     val CDom_sExp = io.fromPreMul.sExpSum - io.fromPreMul.doSubMags.zext
[error]                                                                     ^
```
and
```
[error] (run-main-0) java.lang.reflect.InvocationTargetException java.lang.reflect.InvocationTargetException
 ...
Caused by: chisel3.internal.ChiselException: Connection between left (chisel3.core.Bool@669a9) and source (chisel3.core.Bool@66903) failed @: Locally unclear whether Left or Right (both internal)
	at chisel3.internal.throwException$.apply(Error.scala:13)
	at chisel3.core.Data.bulkConnect(Data.scala:222)
	at chisel3.core.Data.$less$greater(Data.scala:273)
	at chisel3.util.Pipe$.apply(Valid.scala:40)
	at chisel3.util.Pipe$.apply(Valid.scala:46)
	at chisel3.util.Pipe$.apply(Valid.scala:50)
```
A fix is in the works  - #592

ducky64, 2017-04-28:
592 is merged, is this issue resolved?

ucbjrl, 2017-04-28:
Closed by #592

closed 2017-04-28 by ucbjrl