chipsalliance/chisel #594: Multi clock definition

chiselmonster, 2017-04-26:
Hi All,

I am trying to create multiple clocks, but I am unable to find the correct syntax for the new RegInit Command

```
val clock_b_reg = RegInit(UInt(8.W), 0.U(), clock = io.clockB)
```

This syntax is incorrect, any advice on how to create a multi clock design?

Thanks

ducky64, 2017-04-26:
Constructs in Chisel do not take clocks as an argument, instead, you scope them in a `withClock` or `withClockAndReset` block. For example,
```scala
// elements here clocked against module clock
withClock (io.clockB) {
  // elements clocked against io.clockB in this withClock scope
  val clock_b_reg = RegInit(UInt(8.W), 0.U())
}
```

chiselmonster, 2017-04-26:
Hi,

Thanks for the prompt reply, but I still have problem defining clockB

I defined it as 

```
    val clockB = Clock
```
When I tried to use withClock

It still says type mismatch, its expecting type Clock but I am giving it Clock.type

Thanks

edwardcwang, 2017-04-26:
`val clockB = Input(Clock())`

chiselmonster, 2017-04-26:
Thanks man! Worked like magic!! :)

edwardcwang, 2017-04-26:
https://github.com/ucb-bar/chisel3/wiki/Multiple-Clock-Domains

closed 2017-04-26 by edwardcwang

chiselmonster, 2017-04-27:
Thank you so much!

chiselmonster, 2017-04-27:
Hi Edward,

Sorry I still face issues, do I need to define clockB externally? If not how will Chisel know the relationship between the implicit clock and the defined clockB?

I tried to create a class Clock as per the Chisel manual but got stuck after that

I see this error

```

[info] Running PWMIN
[info] [0.000] Elaborating design...
[error] (run-main-0) java.lang.NullPointerException
java.lang.NullPointerException
        at PWMIN$$anonfun$44.apply(PWMIN.scala:63)
        at PWMIN$$anonfun$44.apply(PWMIN.scala:60)
        at chisel3.core.withClockAndReset$.apply(MultiClock.scala:26)
        at chisel3.core.withClock$.apply(MultiClock.scala:41)
        at PWMIN.<init>(PWMIN.scala:60)
        at PWMIN$$anonfun$main$1.apply(PWMIN.scala:417)
        at PWMIN$$anonfun$main$1.apply(PWMIN.scala:417)
        at chisel3.core.Module$.do_apply(Module.scala:42)
        at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:91)
        at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:91)
        at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:236)
        at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:234)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
        at chisel3.internal.Builder$.build(Builder.scala:234)
        at chisel3.Driver$.elaborate(Driver.scala:91)
        at chisel3.Driver$.execute(Driver.scala:126)
        at chisel3.iotesters.setupFirrtlTerpBackend$.apply(FirrtlTerpBackend.sca
la:110)
        at chisel3.iotesters.Driver$$anonfun$execute$1.apply$mcZ$sp(Driver.scala
:48)
        at chisel3.iotesters.Driver$$anonfun$execute$1.apply(Driver.scala:35)
        at chisel3.iotesters.Driver$$anonfun$execute$1.apply(Driver.scala:35)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
        at chisel3.iotesters.Driver$.execute(Driver.scala:35)
        at PWMIN$.main(PWMIN.scala:417)
        at PWMIN.main(PWMIN.scala)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.lang.reflect.Method.invoke(Unknown Source)
[trace] Stack trace suppressed: run last compile:run for the full output.
java.lang.RuntimeException: Nonzero exit code: 1
        at scala.sys.package$.error(package.scala:27)
[trace] Stack trace suppressed: run last compile:run for the full output.
[error] (compile:run) Nonzero exit code: 1
[error] Total time: 1 s, completed 27 Apr, 2017 10:10:45 AM
```
Any advise?

Thanks again

jackkoenig, 2017-04-27:
Can we see your code? A NullPointerException definitely should not be happening. 

What clocking relationship are you trying to create and what are you expecting Chisel to do with that information? Mutli-clock support in Chisel is still relatively young, and Chisel does not handle the subtleties of logic/muxing clocks. Instead we delegate that effort to the designer and backend EDA tools.

[rocket-chip](https://github.com/ucb-bar/rocket-chip) is a proper multi-clock Chisel project that has the [right primitives](https://github.com/ucb-bar/rocket-chip/blob/master/src/main/scala/util/Crossing.scala) to handle this. However, there is the big caveat that rocket-chip assumes clocks are generated externally and thus the designer provides the constraints on these clocks to the EDA tools. 

A future feature we would like to provide is *some* help automating the generation of this constraint information (see https://github.com/ucb-bar/firrtl/issues/537), but we don't have anything like this today.

shunshou, 2017-04-27:
Here's an example of a divide by n (non 50% duty cycle) clk gen I made. There's a lot of other bells and whistles that you should probably ignore and that I will eventually change...

https://github.com/ucb-bar/barstools/blob/master/tapeout/src/main/scala/transforms/clkgen/ClkDivider.scala

Note that I had to make a Verilog black box model of an asynchronously reset register.

chiselmonster, 2017-04-28:
Hi Jackkoenig,
I believe its something to do with my code sequence, the NullPointerException is resolved somehow
I am making a power management controller core which will require a new slow clocks for counting. The main core is running on the fast clock, which is where the cross clock domain will be.
Yes, all the external clocks will be provided externally by DLLs or crystals and I will normally add these constraints in my synthesis script
Thanks for the rocket-chip link, I will read through it but since I am still relatively new to Chisel, I might have additional questions soon, haha

chiselmonster, 2017-04-28:
Hi Shunshou,

Thanks for the link, will check it out asap