chipsalliance/chisel #596: Unresolved dependency on sbt compile

derim422, 2017-04-27:
With the current commit (36718cf6040990f2be9ab143adb1d3c519e9d983), I'm getting an unresolved dependency when running sbt compile. Am I missing something locally that isn't covered in the instructions?

Warning message received:

[warn] 	module not found: edu.berkeley.cs#firrtl_2.11;1.1-SNAPSHOT
[warn] ==== local: tried
[warn]   /home/<redacted>/.ivy2/local/edu.berkeley.cs/firrtl_2.11/1.1-SNAPSHOT/ivys/ivy.xml
[warn] ==== public: tried
[warn]   https://repo1.maven.org/maven2/edu/berkeley/cs/firrtl_2.11/1.1-SNAPSHOT/firrtl_2.11-1.1-SNAPSHOT.pom
[warn] ==== local-preloaded-ivy: tried
[warn]   /home/<redacted>/.sbt/preloaded/edu.berkeley.cs/firrtl_2.11/1.1-SNAPSHOT/ivys/ivy.xml
[warn] ==== local-preloaded: tried
[warn]   file:////home/<redacted>/.sbt/preloaded/edu/berkeley/cs/firrtl_2.11/1.1-SNAPSHOT/firrtl_2.11-1.1-SNAPSHOT.pom
[info] Resolving jline#jline;2.12.1 ...

ucbjrl, 2017-04-27:
The dependency on firrtl 1.1-SNAPSHOT hasn't changed since it was introduced, over 7 months ago. If you're using the master branch of chisel3, you'll need to clone and `sbt publish-local` [firrtl](https://github.com/ucb-bar/firrtl). If you've successfully built chisel3 in from the master branch within the last 7 months, you must have cloned and locally published firrtl. Did you clean out your `~/.ivy2` directory?

The [release branch of chisel3](https://github.com/ucb-bar/chisel3/tree/release) uses the published version of firrtl so the local clone and publish step isn't required.

We'll update the README to reflect this.

derim422, 2017-04-27:
That did it. checked out the release branch and it compiled successfully.

closed 2017-04-27 by derim422