chipsalliance/chisel #598: Can't find setName method

barrichello91, 2017-04-27:
I recently switched to Chisel 3 and I am trying to change some signal name using the setName method that was available in the previous versions of Chisel.

The renaming is necessary to make the signals compatible with AXI4 standard.

What would be the correct way of doing this with Chisel3? I can't find anything on the Wiki.

ducky64, 2017-04-28:
Are you asking about names for internal signals, names for Module ports, or names for black box ports? Can you provide an example of the chisel code you have and what you want the output Verilog to be?

barrichello91, 2017-04-28:
Sorry for the generic question.

I am talking about names for Module ports. I am trying to have my module port interfacing with AXI4 Master within Xilinx Vivado. Years ago I used this (https://github.com/maltanar/axi-in-chisel) as a reference to create my signals, however, with Chisel3 the setName function used as here (https://github.com/maltanar/axi-in-chisel/blob/master/AXIDefs.scala) is no more a viable option.

As an example you can see the code attached in here (https://github.com/maltanar/axi-in-chisel/blob/master/AXIDefs.scala). This process of changing signal name is made as in this way, Xilinx Vivado directly recognizes signals when trying to import an Verilog code dealing with AXI4 Interfaces. 

Is there another way of doing this?

Thanks for your help

Martoni, 2017-04-28:
Hi @barrichello91,
I tryied exactly the same things with the same project yesterdays ;)
The main problem is that signals are defined in a library with a class Bundle. But specific name must be set when it's used in Module ports.
I saw that we can define signals names without the «io_» prefix using [RawModule](https://github.com/ucb-bar/chisel3/wiki/Multiple-Clock-Domains) but how can we define it under the library ?

ducky64, 2017-04-28:
Ah, I see.

So currently, not really. While Module ports are named by the name given the the variable, they do have the `io_` prefix, because the top-level IO bundle is named `io`.

In the `chisel3.experimental` package, there is a `MultiIOModule` which doesn't require a top-level IO bundle named `io`, and allows multiple top-level data.

Consider trying:
```scala
import chisel3.experimental.MultiIOModule

class MyAxiThingy extends MultiIOModule {
  val M = IO(new Bundle {
    val AXI = new Bundle {
      val AWADDR = UInt(...)
      val AWPROT = UInt(...)
    }
  })
}
```
Chisel naming inserts a underscore between sub-bundle elements, which is why this works.
However, since this is an experimental feature, it is subject to change or removal. While this feature gives better control over naming, some are iffy over not having all IOs centralized in `io.`.

ducky64, 2017-04-28:
Also, the reason that we got rid of methods like setName and addPin is that mutable state is in general dangerous; you do a setName that gets stomped by a future setName operation, this sets the stage for obscure and annoying bugs.

Suggestions for other naming APIs are appreciated if you have ideas.

edwardcwang, 2017-04-28:
What's wrong with just RawModule?

ducky64, 2017-04-28:
RawModule works, though it doesn't have the implicit clock and reset. Use RawModule if you don't want the implicit clock and reset, MultiIOModule otherwise.

ducky64, 2017-11-08:
Solved with RawModule. Further discussion on whether we should have a mutable setName or a adapter wrapper constructor should be a separate issue.

closed 2017-11-08 by ducky64