chipsalliance/chisel #600: Chisel2 bulk-connects are not longer supported in chisel3?

ckdur, 2017-04-28:
Greetings.

I'm having the next issue (trying to compile Zscale using chisel3).

ctrl.scala
```
class MemIO(implicit p: Parameters) extends Bundle {
    val imem = new HastiMasterIO()(p.alterPartial({
      case HastiId => "00000"
      case HastiKey("00000") => 
        HastiParameters(
          dataBits=p(XLen),
          addrBits=p(XLen)
        )
    }))
    val dmem = new HastiMasterIO()(p.alterPartial({
      case HastiId => "00001"
      case HastiKey("00001") => 
        HastiParameters(
          dataBits=p(XLen),
          addrBits=p(XLen)
        )
    }))
}
```

zscale.scala

```
class Zscale(resetSignal: Bool = null)(implicit val p: Parameters) extends Module(_reset = resetSignal)
    with HasZscaleParameters {
  val io = new Bundle {
    val mem = new MemIO
    //val prci = new PRCITileIO().flip
  }

  val ctrl = Module(new Control)
  val dpath = Module(new Datapath)

  // TODO: Those bulk connections are incompatible
  io.mem.imem <> ctrl.io.mem.imem
  io.mem.imem <> dpath.io.mem.imem
  io.mem.dmem <> ctrl.io.mem.dmem
  io.mem.dmem <> dpath.io.mem.dmem
  dpath.io.ctrl <> ctrl.io.dpath

  //ctrl.io.prci <> io.prci
  //dpath.io.prci <> io.prci
}
```

The thing here is that certain ports of imem/dmem are assigned either in ctrl or dpath. Expected result is to connect the only ones are used in each module, but instead are only connecting dpath ones:

zscale.DefaultConfig.v
 ```
 assign io_mem_imem_htrans = dpath_io_mem_imem_htrans;
  assign io_mem_imem_hmastlock = dpath_io_mem_imem_hmastlock;
  assign io_mem_imem_haddr = dpath_io_mem_imem_haddr;
  assign io_mem_imem_hwrite = dpath_io_mem_imem_hwrite;
  assign io_mem_imem_hburst = dpath_io_mem_imem_hburst;
  assign io_mem_imem_hsize = dpath_io_mem_imem_hsize;
  assign io_mem_imem_hprot = dpath_io_mem_imem_hprot;
  assign io_mem_imem_hwdata = dpath_io_mem_imem_hwdata;
  assign io_mem_dmem_htrans = dpath_io_mem_dmem_htrans;
  assign io_mem_dmem_hmastlock = dpath_io_mem_dmem_hmastlock;
  assign io_mem_dmem_haddr = dpath_io_mem_dmem_haddr;
  assign io_mem_dmem_hwrite = dpath_io_mem_dmem_hwrite;
  assign io_mem_dmem_hburst = dpath_io_mem_dmem_hburst;
  assign io_mem_dmem_hsize = dpath_io_mem_dmem_hsize;
  assign io_mem_dmem_hprot = dpath_io_mem_dmem_hprot;
  assign io_mem_dmem_hwdata = dpath_io_mem_dmem_hwdata;
```

You can find my advances inside https://github.com/onchipuis/zscale/tree/debug if you want.
There is no way to do this? I need to do all memory connections inside certain module then pass values using aditional ports between modules in order to preserve only 1 module to do the connections

Thanks in advice.

colinschmidt, 2017-04-28:
In chisel3 you need to break it down to the subfields you want to connect. So if only parts of imem connect to parts of ctrl you should use bulk connect on those parts. Perhaps you need smaller subbundles to make this easier

ckdur, 2017-04-28:
Well, is impossible to do this as Zscale's dmem/imem ports are both declared and used inside Control and Datapath Bundles. Subbundles are not possible either because I'm using the HastiMasterIO bundles from rocket-chip, there is no way to extract subbundles from here (actually is possible but the purpose is this code to be compatible with rocket-chip repository).

ducky64, 2017-11-08:
@colinschmidt is this issue still relevant? And is this a chisel core issue, or a zscale issue?

closed 2017-11-08 by ducky64

colinschmidt, 2017-11-08:
This is just a code organization issue as far as I'm concerned. (so not relevant)