chipsalliance/chisel #603: Bulk-connect operator gets confused with Wire

edwardcwang, 2017-04-30:
Say if we had

```scala
  val test = counter === 0.U(3.W)
  val test2 = Wire(0.U(3.W))
  test <> test2
```

causes

```
[info]   chisel3.internal.ChiselException: Connection between left (chisel3.core.Bool@44) and source (chisel3.core.UInt@46) failed @: Locally unclear whether Left or Right (both internal)
[info]   at chisel3.internal.throwException$.apply(Error.scala:13)
[info]   at chisel3.core.Data.bulkConnect(Data.scala:215)
[info]   at chisel3.core.Data.$less$greater(Data.scala:266)
```

Shouldn't this be treated as equivalent to the following, since it wouldn't make sense to assign to test?

```scala
  val test = counter === 0.U(3.W)
  val test2 = Wire(0.U(3.W))
  test2 := test
```

sdtwigg, 2017-05-01:
Maybe, it could be done.

Although this kind of goes under the umbrella of: bulk-connect (as it is now) should generally only be used on I/Os because it is simpler to legislate the connectivity and you don't have to remember the particulars of what kind of internal node it is. Similarly, this would only be helpful for hardware types that are known mono-directional (since ops are the only thing that return an internal read-only signal). Yet, in those cases, the mono-connect operator is perfectly sufficient and, honestly, clearer.

Also, am afraid of the user now needing to become aware of internal implementation details.
For example asTypeOf (and the perennially confusing Vec.apply(Seq)) methods can return a Wire.
```scala
val myWire = Wire(Vec(16, Bool())
myWire <> 0.asTypeOf(myWire)
```
will fail because you are trying to connect two wires.

ducky64, 2017-11-08:
This seems to be case where MonoConnect should be used. If there's a more complex / compelling case, open another issue.
This goes into the larger issues of whether BulkConnect should be smarter about direction inference.

ducky64, 2017-11-08:
Resolution: Update documentation so that BulkConnect should only be used with directioned things, and is not magical

edwardcwang, 2017-12-13:
Fixed in https://github.com/freechipsproject/chisel3/wiki/Interfaces-Bulk-Connections#bulk-connections

closed 2017-12-13 by edwardcwang