chipsalliance/chisel #606: Bundles with structurally typed fields in compatibility mode fail in Scala 2.12

ucbjrl, 2017-05-03:
Compiled with Scala 2.12, the following code:
```
    import Chisel._
    val myReset = true.B
    class ModuleExplicitReset(reset: Bool) extends Module(_reset = reset) {
      val io = new Bundle {
        val done = Bool(OUTPUT)
      }
      io.done := false.B
    }
```
produces
```
Error:(14, 10) value done is not a member of Chisel.Bundle
      io.done := false.B
```
Wrapping the `io` definition in `IO()` or extending `chisel3.core.UserModule` instead of `chisel3.core.LegacyModule`eliminates the error. It seems to be a property of the `LegacyModule` definition.

kammoh, 2017-05-03:
This is actually coming from a deeper issue with recent versions of the Scala compiler, going back at least to 2.12.0-M5 and also present in dotty, and actually hast nothing to do with Chisel:
`scala> trait B {
     | }
defined trait B

scala> 

scala> trait BaseClass {
     |   val x : B
     | }
defined trait BaseClass

scala> 

scala> class MyClass extends BaseClass {
     |   val x = new B {
     |     val b = 0
     |   }
     | 
     |   x.b
     | }
<console>:18: error: value b is not a member of B
         x.b
           ^
`
It's probably introduced by [#5141](https://github.com/scala/scala/pull/5141) or maybe even [#5294](https://github.com/scala/scala/pull/5294) "Inferred types for fields" introduced in the 2.12 branch.

It's quite interesting that the IO() macro somehow gets through with this issue.

jackkoenig, 2017-08-17:
As mentioned here https://github.com/scala/bug/issues/10047 and in "Inferred types for fields" on the 2.12.0 release notes (http://scala-lang.org/news/2.12.0/), this is intended behavior.

Type inference has changed so the type of a structurally typed val is inferred to be the parent type rather than the structural type. 

Because `IO(...)` returns the structural type of its argument, the type of the val is inferred to be the structural type:
```scala
scala> trait A
defined trait A

scala> abstract class B {
     |   val a: A
     | }
defined class B

scala> class C extends B {
     |   def f[T <: A](a: T): a.type = a
     |   val a = f(new A {
     |     val foo = 3
     |   })
     | }
defined class C

scala> val c = new C
c: C = C@72b59b59

scala> c.a.foo
res0: Int = 3
```

This can be worked around with the scalac option `-Xsource:2.11`.

Assuming this workaround is general enough for large codebases like rocket chip, I think we should move forward with bumping to 2.12.3 and use the workaround for now. `chisel3._` code should not be affected.

ducky64, 2017-08-17:
That plan of action sounds reasonable. Interesting that wrapping it in an IO(...) call is a workaround, is that something Scala might also remove in the future?

ducky64, 2017-09-22:
Is this still relevant? It appears there's really nothing we can do on our end, and the IO(...) seems to solve this problem for relevant use cases. Compatibility code (rocket-chip) is basically the only issue, for which there is the described scalac option workaround.

jackkoenig, 2017-09-22:
I think given that it's solved by `IO(...)` this issue is closed from our perspective.

To reiterate for posterity: Users of the compatibility wrapper should add the scalac option `-Xsource:2.11`

ucbjrl, 2017-09-22:
We mention this in the release notes.

wachag, 2017-11-28:
In the (very far...) future this problem will come again when Dotty is released.

jackkoenig, 2018-08-24:
This is a problem again (and has been for a while now). https://github.com/freechipsproject/chisel3/pull/754 changed the type definition of `def IO`:
```scala
// From
protected def IO[T<:Data](iodef: T): iodef.type
// To
protected def IO[T<:Data](iodef: T): T 
```
Returning `iodef.type` works around this issue, returning `T` does not.

azidar, 2020-10-27:
This will be fixed in 3.5.0 once we delete `val io`.

jackkoenig, 2022-03-22:
Fixed in 3.5.0

closed 2022-03-22 by jackkoenig