chipsalliance/chisel #608: SyncReadMem strage read behavior when rdEn gets deasserted

aayupov, 2017-05-04:
I am seeing that the read data output port gets an X (random) value in FIRRTL interpreter when I deassert rdEn in the previous cycle whereas I would expect the port to keep the previous read data instead. In fact, the generated verilog behaves as I expected. Do you think it is a bug or this is intentional? As far as I know all sequential memories behave as I would like here.

aswaterman, 2017-05-04:
ASIC SRAMs, which SyncReadMem needs to model, generally have undefined outputs after the read-enable is not asserted.

ducky64, 2017-11-08:
Perhaps make consistent with behavior in #550

ducky64, 2017-12-13:
Behavior is defined to undefined, which is the correct behavior when modeling ASIC SRAMs
Workaround: explicitly register the data.

closed 2017-12-13 by ducky64