chipsalliance/chisel #612: setName equivalent in Chisel3

ssbanerje, 2017-05-16:
I want to understand how to rename signals for module ports in Chisel3 generated verilog code. An example of what I am trying to do can be found on slide 6 on [link](https://inst.eecs.berkeley.edu/~cs250/sp16/disc/Blackboxing.pdf).

As I understand in older versions of Chisel this could be done using the `setName` function. Is there similar functionality in Chisel3.

This is sort of related to #598. However the main difference between my problem and the solution presented there in terms of the `MultiIOModule` is that my external IP does not follow a similar hierarchical naming scheme using the `_` character. Could I directly set the strings to be used in the generated verilog?

jackkoenig, 2017-05-16:
Using MultiIOModule could you do something like this?

```scala
class VecMultiIOModule extends MultiIOModule {
  val ap_clk = IO(Input(Bool())
  val ap_rst = IO(Input(Bool()))
  val ap_start = IO(Input(Bool()))
  val ap_done = IO(Output(Bool()))
  // ... or whatever names you want
```

ssbanerje, 2017-05-16:
I have a bunch of bundles which are logically grouped together inside my Chisel code. Something like
```scala
class MMIOInterface extends Bundle {
  // MMIO Valid
  val ha_mmval = Input(Bool())
  // MMIO AFU desc. space
  val ha_mmcfg = Input(Bool())
  // MMIO read/write
  val ha_mmrnw = Input(Bool())
  // MMIO wordsize
  val ha_mmdw = Input(Bool())
  // MMIO address
  val ha_mmad = Input(Bits(24.W))
  // MMIO address parity
  val ha_mmadpar = Input(Bool())
  // MMIO write data
  val ha_mmdata = Input(Bits(64.W))
  // MMIO write data parity
  val ha_mmdatapar = Input(Bool())
  // MMIO write ack/read valid
  val ah_mmack = Output(Bool())
  // MMIO read data
  val ah_mmdata = Output(Bits(64.W))
  // MMIO read data parity
  val ah_mmdatapar = Output(Bool())
}

class ControlInterface extends Bundle {
  // Control valid
  val ha_jval = Input(Bool())
  // Control command
  val ha_jcom = Input(Bits(8.W))
  // Control command parity
  val ha_jcompar = Input(Bool())
  // Control wed address
  val ha_jea = Input(Bits(64.W))
  // Control wed parity
  val ha_jeapar = Input(Bool())
  // Control AFU running
  val ah_jrunning = Output(Bool())
  // Control AFU done
  val ah_jdone = Output(Bool())
  // Control LLCMD ACK
  val ah_jcack = Output(Bool())
  // Control AFU error
  val ah_jerror = Output(Bits(64.W))
  // Control yield
  val ah_jyield = Output(Bool())
  // Control timebase request
  val ah_tbreq = Output(Bool())
  // Control AFU parity support
  val ah_paren = Output(Bool())
  // Control AFU clock
  val ha_pclock = Input(Clock())
}
```

And these are used inside modules that take each bundle as a port definition. However my top-level module that interfaces with an external verilog module must use the same wires in a flat namespace. i.e.,

```vhdl
LIBRARY ieee;
USE ieee.std_logic_1164.ALL;
USE ieee.numeric_std.ALL;

ENTITY afu IS
PORT
(
...
-- MMIO Interface
  ha_mmval      :  IN STD_LOGIC;                      -- MMIO valid
  ha_mmcfg      :  IN STD_LOGIC;                      -- MMIO AFU desc. space
  ha_mmrnw      :  IN STD_LOGIC;                      -- MMIO read/write
  ha_mmdw       :  IN STD_LOGIC;                      -- MMIO wordsize
  ha_mmad       :  IN STD_LOGIC_VECTOR(23 DOWNTO 0);  -- MMIO address
  ha_mmadpar    :  IN STD_LOGIC;                      -- MMIO address parity
  ha_mmdata     :  IN STD_LOGIC_VECTOR(63 DOWNTO 0);  -- MMIO write data
  ha_mmdatapar  :  IN STD_LOGIC;                      -- MMIO write data parity
  ah_mmack      : OUT STD_LOGIC;                      -- MMIO write ack/read valid
  ah_mmdata     : OUT STD_LOGIC_VECTOR(63 DOWNTO 0);  -- MMIO read data
  ah_mmdatapar  : OUT STD_LOGIC;                      -- MMIO read data parity
-- Control Interface
  ha_jval       :  IN STD_LOGIC;                      -- Control valid
  ha_jcom       :  IN STD_LOGIC_VECTOR(7 DOWNTO 0);   -- Control command
  ha_jcompar    :  IN STD_LOGIC;                      -- Control command parity
  ha_jea        :  IN STD_LOGIC_VECTOR(63 DOWNTO 0);  -- Control wed address
  ha_jeapar     :  IN STD_LOGIC;                      -- Control wed parity
  ah_jrunning   : OUT STD_LOGIC;                      -- Control AFU running
  ah_jdone      : OUT STD_LOGIC;                      -- Control AFU done
  ah_jcack      : OUT STD_LOGIC;                      -- Control LLCMD ACK  
  ah_jerror     : OUT STD_LOGIC_VECTOR(63 DOWNTO 0);  -- Control AFU error
  ah_jyield     : OUT STD_LOGIC;                      -- Control yield
  ah_tbreq      : OUT STD_LOGIC;                      -- Control timebase request
  ah_paren      : OUT STD_LOGIC;                      -- Control AFU parity support
  ha_pclock     :  IN STD_LOGIC                       -- Control AFU clock
);
END ENTITY;
```

Using `MutiIOModule` will involve redefining every wire as a separate IO port and then inidvidually assigning them to the bundles defined earlier to be passed into the Chisel modules. I think it would be simpler and easier to manintain by just rewriting the names instead of repeating hundreds of wire several times (once in port definition, then assignment to bundles, then in the bundle definitions).

jwright6323, 2018-04-06:
This would be nice to have for BlackBoxes as well, e.g. when you have a 3rd party thing you want to integrate with tweaked names/Chisel types

chick, 2018-12-17:
Won't fix. Use a wrapper. Or propose and RFS

closed 2018-12-17 by chick