chipsalliance/chisel #613: java.lang.OutOfMemoryError: GC overhead limit exceeded

lorenzoditucci, 2017-05-18:
Hello, 

I am trying to generate the Verilog from a design that is composed of multiple Modules, but after some minutes of computation (~10), a java.lang.OutOfMemoryError: GC overhead limit exceeded exception is thrown.

Reducing the dimension for the input of the modules chisel is able to generate Verilog, is there any workaround to this issue?

Thanks

jackkoenig, 2017-05-18:
You can increase the java heap size. If you are using sbt, you can try `sbt -mem 4096` (or some other amount)

lorenzoditucci, 2017-05-18:
Thanks, I managed to generate the Verilog using: 
`export _JAVA_OPTIONS='-XX:-UseGCOverheadLimit -Xmx10g '`

I was wondering, is this the correct behavior, it is a problem of my design that it is too complicated or it is something that depends on the "translator"?

jackkoenig, 2017-05-18:
Larger designs require more heap space. It's not really a "problem", it just seems that the JVM decides not to allocate all that much space by default. We could probably be more efficient with memory in chisel and firrtl but there's still the fundamental issue that larger designs will always need more heap space.

closed 2017-05-18 by jackkoenig