chipsalliance/chisel #621: Behavior of register or wires that is never assigned

jchang0, 2017-06-01:
What should be the behavior of a declared register that is never assigned?

If I declare a register that is never assigned to anything [e.g.  val ran_val_reg = Reg(Bool())  ], I would imagine that it should clock in a random value every clock cycle.

However, it looks like chisel optimize it away into a wire and assign it to a random value during initialization.  This would have a different behavior than having a random value every clock cycle.

This is also related to this:
https://github.com/ucb-bar/chisel3/issues/576

jackkoenig, 2017-06-01:
The default next value of a register is the register itself, so the behavior you are seeing is the defined behavior. The initial randomization is due to the fact that the register has no reset.

Using the register itself as the default "next" value is what allows you to write such things as:
```scala
val myReg = Reg(UInt(8.W))
when (someCondition) {
  myReg := 12.U
}
// When someCondition is false, myReg doesn't change
```

For random numbers at runtime, Chisel has [LFSR16](https://chisel.eecs.berkeley.edu/api/index.html#chisel3.util.LFSR16$)

jchang0, 2017-06-01:
Ah, then the other bug I filed is also a bug here then. Because chisel optimizes away the register into a wire, so in formal verification (and in synthesis), the wire becomes dangling and can change value from cycle to cycle, as opposed to the expected behavior of having a constant random value.

jchang0, 2017-06-01:
Actually, this causes simulation-synthesis mismatch on any registers whose input is unspecified (or in this particular case I'm looking at--optimized away)

I'm seeing a circuit in rocketchip for a very strip down version.  The firrtl file have the register:
    reg reg_entries : UInt<54>[5], clock @[TLB.scala 84:24]
   ....
      reg_entries[_T_1083] <= _T_1138 @[TLB.scala 168:24]

But when I looked at the verilog file, it no longer have the T_1138 signal, and reg_entries doesn't have the always @(posedge clock) block surrounding reg_entries any longer.

Even if T_1138 signal is optimized away, shouldn't it still keep the register since the expected behavior is that this register value should not change from cycle to next cycle?  This is the simulated behavior, but it seems like it wouldn't be the case for synthesized circuit from this verilog code.

jackkoenig, 2017-06-01:
Interestingly, it doesn't look like the Firrtl optimization passes are doing this, it looks like the Firrtl Verilog emitter is doing it. But to illustrate the bug @jchang0 is pointing out:

Chisel:
```scala
class RandTop extends Module {
  val io = IO(new Bundle {
    val out = Output(UInt(32.W))
  })
  val rand = Reg(UInt(32.W))
  io.out := rand
}
```
Firrtl:
```
circuit RandTop :
  module RandTop :
    input clock : Clock
    input reset : UInt<1>
    output io : {out : UInt<32>}

    clock is invalid
    reset is invalid
    io is invalid
    reg rand : UInt<32>, clock @[Rand.scala 8:17]
    io.out <= rand @[Rand.scala 9:10]
```
Verilog:
```verilog
module RandTop(
  input         clock,
  input         reset,
  output [31:0] io_out
);
  reg [31:0] rand$;
  reg [31:0] _GEN_0;
  assign io_out = rand$;
`ifdef RANDOMIZE
  integer initvar;
  initial begin
    `ifndef verilator
      #0.002 begin end
    `endif
  `ifdef RANDOMIZE_REG_INIT
  _GEN_0 = {1{$random}};
  rand$ = _GEN_0[31:0];
  `endif
  end
`endif
endmodule
```

And to make sure I understand, what we want is something like this:
```verilog
module RandTop(
  input         clock,
  input         reset,
  output [31:0] io_out
);
  reg [31:0] rand$;
  reg [31:0] _GEN_0;
  assign io_out = rand$;
`ifdef RANDOMIZE
  integer initvar;
  initial begin
    `ifndef verilator
      #0.002 begin end
    `endif
  `ifdef RANDOMIZE_REG_INIT
  _GEN_0 = {1{$random}};
  rand$ = _GEN_0[31:0];
  `endif
  end
`endif
  always @(posedge clock) begin
    rand$ <= rand$;
    // or perhaps
    rand$ <= _GEN_0;
endmodule
```

jchang0, 2017-06-01:
Right, we want something like this:

  always @(posedge clock) begin
    if (reset) begin
      rand$ <= _GEN_0;
    end else begin
      rand$ <= rand$;
    end
  
Before I thought mine was a special case since one might argue that a uninitialized register that never updates it content is meaningless.

However, it looks like it might be more common than my special usage case, since optimization and parameterization might turn a design with a defined next state into this special case for which the behavior is incorrect.

jackkoenig, 2017-06-01:
Do we really want `rand$` to always get the randomize value on reset though? Really it's randomized when first provide power, since the register doesn't have a reset, it's value should not change on reset, no?

jchang0, 2017-06-01:
Ah.  Yes, that's true.  But we do need a way to initialize it to the random value on power up though instead of leaving it as "X"'s.

jackkoenig, 2017-06-01:
In my above example, so long as `RANDOMIZE_REG_INIT` it should be getting randomized right?

Somewhat unrelated, it seems to be that the default should be to randomize and people should have to manually disable it (since we don't emit X friendly Verilog)...

jchang0, 2017-06-01:
Oh yeah, you are right about the init random values.

aswaterman, 2017-06-01:
The ```always @(posedge clock)``` should contain no assignments, as the register is neither reset nor assigned.

mwachs5, 2017-06-03:
@jackkoenig : "For random numbers at runtime, Chisel has LFSR16"

Unless I am reading the code wrong, LFSR16 is not random at all, even in the pseudo-random sense. 

It is always initialized to 1. Any LFSR16 instantiated in a design would be in sync and different runs of a simulation/real test using it would have the exact same behavior.

I don't think that's a problem (we shouldn't change LFSR16 actually used by hardware), but we shouldn't expect that to be sufficient for a TestHarness which wants to generate something randomly.

closed 2017-06-05 by azidar

jackkoenig, 2017-06-05:
@mwachs5 It looks like you are correct. A proper LFSR library would either not initialize the register or perhaps optionally do it from a seed. But yeah I clearly made incorrect assumptions about LFSR16 in chisel3.util