chipsalliance/chisel #622: Assignment of init random values for register

jchang0, 2017-06-01:
Clifford noted that the way random numbers are generated when number of bits > 32 is not very random:

 _GEN_0 = {2{$random}};

As this will produce one 32-bit random value and then repeat them.  Better to do:
  _GEN_0 = {$random, $random};

In fact, he pointed out that $random isn't very good at all:
http://svn.clifford.at/handicraft/2014/vlogrnd/bad_random.v

I'm not sure what the solution for that is, I suppose the work around is probably different for each simulator.

aswaterman, 2017-06-01:
Can you repost on Firrtl?  This is a deficiency in the Verilog emitter.

jchang0, 2017-06-01:
Migrated to

https://github.com/ucb-bar/firrtl/issues/598

closed 2017-06-01 by jchang0