chipsalliance/chisel #624: Can not install the sbt

okbonhahaha, 2017-06-03:
Hi ~ hope that someone could help with this issue

I am using Ubuntu 16.04.
I try to install the sbt tool from the following instructions described on the scala-sbt official website

_echo "deb https://dl.bintray.com/sbt/debian /" | sudo tee -a /etc/apt/sources.list.d/sbt.list
sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv 2EE0EA64E40A89B84B2DF73499E82A75642AC823
sudo apt-get update
sudo apt-get install sbt_

but at the step of "sudo apt-get install sbt" , I get the following error message

**/usr/share/sbt/bin/sbt-launch-lib.bash: line 73: java: command not found
dpkg: error processing package sbt (--configure):
 subprocess installed post-installation script returned error exit status 127
Errors were encountered while processing:
 sbt
E: Sub-process /usr/bin/dpkg returned an error code (1)**

Is there anyone encounter the same issue? How do you solve it? 

Thanks a lot!

edwardcwang, 2017-06-03:
Do you have Java installed? https://www.digitalocean.com/community/tutorials/how-to-install-java-with-apt-get-on-ubuntu-16-04

okbonhahaha, 2017-06-06:
@edwardcwang 
Thanks a lot! It seems that I have to Java first!!

closed 2017-06-06 by okbonhahaha