chipsalliance/chisel #626: When attempting to compile: `build.sbt:43: error: illegal start of simple expression`

gmarkall, 2017-06-12:
I'm attempting to follow the directions in the README to build Chisel3 on Fedora 25. I have Verilator and sbt installed. When I run `sbt compile` I get the following error:

```
/home/graham/work/projects/rocket-chip-gen/chisel3/build.sbt:43: error: illegal start of simple expression
  name := "chisel3",
                    ^
sbt.compiler.EvalException: Error parsing definition.  Ensure that there are no blank lines within a definition.
        at sbt.compiler.Eval.checkError(Eval.scala:343)
        at sbt.compiler.Eval.sbt$compiler$Eval$$parse(Eval.scala:268)

...

[error] sbt.compiler.EvalException: Error parsing definition.  Ensure that there are no blank lines within a definition.
[error] Use 'last' for the full log.
Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? q
```

I'm guessing that something fairly simple is wrong with my setup, but I'm not familiar with Scala or sbt, so I'm struggling to see how to troubleshoot this problem. Any advice or thoughts on what to try would be much appreciated - many thanks in advance!

ucbjrl, 2017-06-12:
This looks like an `sbt` version issue, coupled with inconsistent style in chisel's `build.sbt`. We'll fix the latter, but in the meantime, you might ensure you're using a _modern_ version of `sbt`.
```
sbt sbt-version
```
should give you the version of `sbt` you're using, but I'm afraid that may not work if `sbt` is unable to correctly parse the `build.sbt` file.
If that should fail with a similar error, try the following sequence:
```
mkdir /tmp/tsbt
cd /tmp/tsbt
sbt -sbt-create sbt-version
```
This tacitly assumes that your `sbt` is recent enough to understand those options.

Alternatively, `sbt` is typically a shell script that invokes `Java` with the appropriate sbt jar file. If
```
file `which sbt`
```
indicates `sbt` is a shell script, you might examine it for version information:
```
grep version `which sbt`
```
Any `sbt` version from `0.13.11` to `0.13.15` (the current release) should work. If your installed version of `sbt` is older than that, you should download and install the 0.13.15 version from http://www.scala-sbt.org/download.html

gmarkall, 2017-06-12:
Thanks for the response - it looks like I have an old version:

```
$ sbt sbt-version
Getting org.scala-sbt sbt 0.13.1 ...
:: retrieving :: org.scala-sbt#boot-app
        confs: [default]
        41 artifacts copied, 0 already retrieved (12124kB/52ms)
[info] Set current project to graham (in build file:/home/graham/)
[info] 0.13.1
```

I'll try the latest version - I think I was a bit confused because SBT appeared to download 0.13.15 (presumably as it is referenced in `project/build.properties`) so I thought I was getting the latest version automatically already. 

Thanks for the help!

closed 2017-06-12 by edwardcwang