chipsalliance/chisel #630: Any knobs to disable the optimization?

sxu55, 2017-06-19:
I noticed that the more recent version of Chisel 3 releases (e.g. 05-30) start to optimize out the unused ports of non-toplevel modules, while older vers (e.g. 02-24) doesn't. We sometimes need to do post-verilog manual connection of those unused ports in non-toplevel modules, and cannot find a knob to disable this optimization during Chisel compilation. Is there a option provided to disable such optimization?

jackkoenig, 2017-06-19:
I apologize for the optimization making it into release without the adequate APIs for use cases like yours. We have a release coming on Wednesday that adds the `dontTouch(...)` API where you can mark signals as something that should not be removed. A simple example of how it could be used:

```scala
import chisel3._
import chisel3.experimental.dontTouch

class ModuleWithDeadCode extends Module {
  val io = IO(new Bundle {
    val foo = Input(UInt(32.W))
    val bar = Output(UInt(32.W))
  })
  io.bar := io.foo

  val myDeadWireConnectedLater = dontTouch(Wire(new SomeBundleType))
  // You could define and mark later:
  val myOtherDeadWire = Wire(UInt(32.W))
  dontTouch(myOtherDeadWire)
}
```

It does live in `chisel3.experimental`, but it should be used. It is possible it could be changed via a deprecation schedule but the feature itself is not going away.

I'm also working on adding the ability to disable the pass in general but this is intended more for debugging purposes and real flows should use `dontTouch`

sxu55, 2017-06-20:
Thanks for the reply! We will try out the new release and see if it works.

jackkoenig, 2017-06-20:
I'm closing this issue as answered but please reopen it if there are problems with the new release!

closed 2017-06-20 by jackkoenig