chipsalliance/chisel #631: Verilator and firrtl simulation gives very different results in the .vcd trace

oharboe, 2017-06-21:
Is there a reason why I should get different results in firrtl and Verilator?

I have been developing/testing with Verilator so far and my code works reasonably well. When I tried to write a unit-test with firrtl, the same test that passed in Verilator failed in firrtl.

Notice in the firrtl trace:

- The width of the reset signal for the sub-module is half the width of that of the parent module
- Signals in firrtl default to undefined. I guess this is intentional?
- The sub.io.enable signal goes to 0 half a clock *before* io.enable goes to 0 in the parent module

firrtl vcd file:

[image]

Verilator vcd file:

[image]

[vcdfiles.zip](https://github.com/freechipsproject/chisel3/files/1091774/vcdfiles.zip)

chick, 2017-06-22:
Any chance you could post the circuit under test. 
Some work was done to try and clean this up in firrtl-interpreter [aef7f93](https://github.com/freechipsproject/firrtl-interpreter/commit/aef7f934b1dd28672ce4a3b8e8c913724100e8a8) can you check and see if you are using an interpreter with this commit?

oharboe, 2017-06-22:
it is in the zip file in the issue i logged.

chick, 2017-06-22:
Sorry, I missed that, thanks, I will take a look

oharboe, 2017-06-22:
Note that prompted me to look at this, wasn't that the .vcd traces looked
different, but that a big unit-test that I have works perfectly with the
Verilator backend, then fails with firrtl.

When I look at the .vcd traces, I see that Verilator and firrtl are
completely different.

If I can't trust the .vcd output from firrtl, then figuring out why firrtl
fails and why Verilator works is going to be quite difficult.

I want to use firrtl because it would be faster(shorter startup time) and
doesn't use any subprocesses, which is nice to avoid stuck infinite loop
processes from Verilator piling up on our build servers.

oharboe, 2017-06-23:
I ran into more unit-tests that fail in firttl, but work in Verilator today. Still a pretty big example, but I'll keep an eye out something that I can trim down to an example I can file as an issue.

chick, 2017-06-23:
I'm looking at a fix.  The timing/clock model in the interpreter is vary naive, but continuing to approve.  I would very much like to see any example tests that very between verilator and the interpreter

oharboe, 2017-06-23:
I had a crack at reducing the example, but the problem is that I need to
complete the conversion from Chisel 2 to Chisel 3 of our code first,
otherwise I won't be able to get everything into a single file and peel
away enough that I can release it to the public.

chick, 2017-06-23:
I am still not sure about the problem, but one thing I difference I see is that with verilator pokes that occur before a reset are lost.  For example
```
49  poke(c.io.enable, 1)
50  step(1)
51  poke(c.io.enable, 0)
52  reset(1)
53  poke(c.io.enable, 1)
```
The poke of 0 at line 51 is lost by verilator.  To reduce difference in behavior, try issuing a step before and after reset.  This seems to make the the vcd output more similar in my tests

I don't the verilator behavior makes sense, so I am going to dig some more before trying to make the interpreter in line with it.

oharboe, 2017-06-23:
I don't understand the above traces. Why is it showing reset high for two clocks when you are using reset(1). Similarly, I would have expected the io.enable pulses to be 1 clock wide, not two clocks.

chick, 2017-06-23:
Sorry, I copied the wrong piece of code.  The call was ```reset(2)```.

oharboe, 2017-06-23:
And step(2) as well, then, I guess.

chick, 2017-06-23:
Yes. It was ```step(2)``` as well.

oharboe, 2017-06-23:
Could you share your example code? To me it looks like you might have found another problem.

chick, 2017-06-23:
Here is a gist for it
[Gist: Example of reset and poke difference between verilator and interpreter](https://gist.github.com/chick/3e195883f1785740b553f585af4ba36a)

oharboe, 2017-06-23:
I see.

Idea: add an io.enableOut := io.enable and add an expect(io.enableOut, ...)
throughout....

oharboe, 2017-06-24:
Here's something odd that I can't make sense of in the Verilator trace.

The poke() will change the signal of the falling edge of the clock, whereas e.g. the output of a ShiftRegister() changes the value on the rising edge of a clock.

Given:

	val oMEnaPre = ShiftRegister(blah, 1)
	val oMEna = ShiftRegister(oMEnaPre, 1)

Then the signals should *not* overlap.

[image]

wachag, 2017-06-29:
I have similar issues, verilator and firrtl simulation results are different for some of my test cases too, I'll try to isolate a minimal failing code.

Jerryg6j3, 2017-12-04:
Have encountered the same question @oharboe proposed. The poke signal change at the falling edge while  the circuit is positive edge triggered. Why's that?

oharboe, 2018-10-04:
Attached is a failing unit-test of much the same problem:

[FiddlyBobTests.zip](https://github.com/freechipsproject/chisel3/files/2446209/FiddlyBobTests.zip)

I made a simple chisel.iotesters test of a module that's supposed to output io.b 1 if input io.a is 1 on two consecutive positive clock edges. This is a simplified version of a test of a bigger module. (FiddlyBobTest.scala attached).

As can seen from the wavetrace below, io.b is 1 on a single rising clock edge when io.count == 3 as I intended, but the Chisel unit-tester fails because it is telling me that io.b is 1 on io.count == 2 as well.

What am I missing?

Am I outside the intended scope of chisel.iotesters?

Here's the module in Chisel:

 
class FiddlyBob extends Module {
  val io = IO(new FiddlyBobBundle)
  val previousA = ShiftRegister(io.a, 1)
  io.b := io.a && previousA
}

Here's the Verilog, chopped down for clarity:

module FiddlyBob(
  input         clock,
  input         reset,
  input  [15:0] io_count,
  input         io_a,
  output        io_b
);
  reg  previousA;
  wire  _T_13;
  assign _T_13 = io_a & previousA;
  assign io_b = _T_13;
  always @(posedge clock) begin
    previousA <= io_a;
  end

[image]

 
STARTING test_run_dir/fiddly_bob_tester/VFiddlyBob
[ [35minfo [0m] [0.001] SEED 1
[ [35minfo [0m] [0.003]   POKE FiddlyBob.io_a <- 0
[ [35minfo [0m] [0.003]   POKE FiddlyBob.io_count <- 0
[ [35minfo [0m] [0.003] STEP 0 -> 1
[ [35minfo [0m] [0.011]   PEEK FiddlyBob.io_b -> 0
[ [35minfo [0m] [0.011]   EXPECT FiddlyBob.io_b -> 0 == 0 PASS
[ [35minfo [0m] [0.011]   POKE FiddlyBob.io_count <- 1
[ [35minfo [0m] [0.012] STEP 1 -> 2
[ [35minfo [0m] [0.013]   PEEK FiddlyBob.io_b -> 0
[ [35minfo [0m] [0.013]   EXPECT FiddlyBob.io_b -> 0 == 0 PASS
[ [35minfo [0m] [0.013]   POKE FiddlyBob.io_a <- 1
[ [35minfo [0m] [0.013]   POKE FiddlyBob.io_count <- 2
[ [35minfo [0m] [0.014] STEP 2 -> 3
[ [35minfo [0m] [0.014]   PEEK FiddlyBob.io_b -> 1
[ [35minfo [0m] [0.015] EXPECT AT 3 io.a was 0 on the previous positive clock edge, so it can't be 1 now FAIL
[ [35minfo [0m] [0.015]   PEEK FiddlyBob.io_b -> 1
[ [35minfo [0m] [0.015] io.a was 0 in the previous cycle, so it can't be 1 now  EXPECT FiddlyBob.io_b -> 1 == 0 FAIL
[ [35minfo [0m] [0.016]   POKE FiddlyBob.io_a <- 1
[ [35minfo [0m] [0.016]   POKE FiddlyBob.io_count <- 3
[ [35minfo [0m] [0.016] STEP 3 -> 4
[ [35minfo [0m] [0.017]   PEEK FiddlyBob.io_b -> 1
[ [35minfo [0m] [0.017] io.b should be 1 because io.a has been high for two cycles  EXPECT FiddlyBob.io_b -> 1 == 1 PASS
[ [35minfo [0m] [0.017]   POKE FiddlyBob.io_a <- 0
[ [35minfo [0m] [0.017]   POKE FiddlyBob.io_count <- 4
[ [35minfo [0m] [0.017] STEP 4 -> 5
[ [35minfo [0m] [0.018]   PEEK FiddlyBob.io_b -> 0
[ [35minfo [0m] [0.018] io.b should be high for a single cycle  EXPECT FiddlyBob.io_b -> 0 == 0 PASS
Enabling waves..
Exit Code: 0
[ [35minfo [0m] [0.024] RAN 5 CYCLES FAILED FIRST AT CYCLE 3
- should fiddle the bobs *** FAILED ***
  false was not true (FiddlyBobTests.scala:61)
Run completed in 4 seconds, 178 milliseconds.
Total number of tests run: 1
Suites: completed 1, aborted 0
Tests: succeeded 0, failed 1, canceled 0, ignored 0, pending 0
*** 1 TEST FAILED ***

oharboe, 2018-10-04:
> @oharboe <https://github.com/oharboe>: This may be just an incorrect
> assumption that pokes reset to some default value. Pokes are persistent.
> This test pokes a 1, steps (that 1 gets latched), it's still poking a 1,
> then the expect tries to evaluate 1 && 1 as 0. If the test currently
> wants to do this, then it would need to re-poke a 0 before the expect.
>
> Or, more concretely, pokes are combinational and applied instantly. Also,
> pokes are persistent.
>
> You've actually hit on a common use case and one of the use cases that the
> forthcoming Testers2 proposal attempts to address. Specifically, this adds
> the notion of Duration Scopes where you poke only over some time period
> as well as a weakPoke which provides a fallback default value: #725

Thanks! This helps a lot to clarify.

It seems wrong to poke back to 0 before testing as that just makes the test
work, it's not what I'm trying to test: that io.a being equal two 1 for two
consequtive rising edges cause io.b to go to 1 on the second rising edge.

Near as I can see the master branch of chisel 3 is now Chisel 3.2, in which
Testers2 should be a part. Is there an expected schedule for the Chisel 3.2
release?

seldridge, 2018-10-04:
> that io.a being equal two 1 for two consequtive rising edges cause io.b to go to 1 on the second rising edge.

In this circuit, output `io.b` is combinationally determined by the input `io.a`, so this will assert combinationally as soon as you poke `io.a`. The shift register is just one stage. Consequently, I don't think there's much utility in looking at the behavior across two steps since the internal state can only store one step. What I'm getting at is that `io.b` will assert _before_ the second clock edge.

More concretely, here's a different test showing this:
```scala
// Get the shift register into a known, guaranteed state (0)
poke(c.io.a, 0)
step(1)

// We're still poking c.io.a to 0
expect(c.io.b, 0) // 0 && 0

// Combinationally poke c.io.a to 1
poke(c.io.a, 1)
expect(c.io.b, 0) // 1 && 0

// Now latch the poke making the shift register 1
step(1)
expect(c.io.b, 1) // 1 && 1
```

More concretely in addition to pokes... _peeks and asserts are combinational_.

Testers2 roadmap is a @ducky64 question. My understanding is that this is not upstream and will be split out into a separate repository (like how chisel-testers technically works now). Some stuff is being merged to enable this and there should be something up on Maven for the actual 3.2 release. Test driving this sooner would involve using a special branch of chisel3 or, once the new testers2 repo is populated, I think publishing that locally.

oharboe, 2018-10-04:
This is a good explanation.

To explain my use-case: my design is for an FPGA, so the Chisel code I
wrote will unambigiously have io.b high for a single cycle, which is what I
wanted to test. The code I'm working on is code that I need to put under
test *then* change, so I can't just add a registered output to work around
these problems until I've got it under test.