chipsalliance/chisel #633: Blackbox IO doesn't actually "exist"

ducky64, 2017-06-26:
Related: #249 and #282.

The root cause of both issues is that the BlackBox IO object itself doesn't actually get emitted as FIRRTL, only its children are. So operations on that IO object would become invalid if emitted to FIRRTL, and any properties of it (like flipped-ness) are dropped.

The second is possible to fix by special-case logic in the emitter, and the first is (probably?) already worked around by resolving bulk-connects at the Element (leaf) level in Chisel. But the underlying issue is still there, and if we ever start emitting bulk connects again it will break.

Potential solutions:
- More and more special casing. Very ugly, high likelihood of missed edge cases and bugs.
- Deprecate the "legacy" `BlackBox` interface in favor of `ExtModule`, which supports multiple top-level IO objects and doesn't do this special IO unwrapping. Possibly also introduce an "all IOs" function that returns all the top-level IOs as a record or record-act-alike object. Side note: this may be of interest to `MultiIoModule` users?
- Have the top-level BlackBox IO be dummy object that emits connects and stuff at its immediate children level, instead of being an actual object in the hardware graph. Unclear if this can work without requiring changes to user BlackBox code, since the IO `Bundle` would no longer be a `Bundle`.
- Something else?

ducky64, 2017-07-25:
Resolution: fix #282 with a hack-patch and figure out a long term solution later (either more hack-patching for #249, when chisel3 starts emitting bulk connects again, or pushing towards ExtModule)

ccelio, 2018-04-19:
I believe I am hitting against this issue.  

I am trying to create a `LazyModuleBlackBox`, and the existing `LazyModuleImp`s extend Modules that have `io`s. However, if I attempt to make a LazyBlackBoxModuleImp I get an error that there is no `io` and thus my class must be abstract.

https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/diplomacy/LazyModule.scala#L183-L185

What I would like to do (I think) is:
```
class LazyBlackBoxModuleImp(val wrapper: LazyModule) extends BlackBox with LazyModuleImpLike {
  val (auto, dangles) = instantiate()
}
``