chipsalliance/chisel #634: poke(c.io.somesignal, 0xffffffff) works with Verilator, but fails with firrtl

oharboe, 2017-06-27:
The workaround is to use  poke(c.io.PNewInst, BigInt("ffffffff", 16))

chick, 2017-06-27:
Can you provide the declaration of c.io.PNewInst

oharboe, 2017-06-27:
Input(UInt(32.W))

chick, 2017-06-27:
Another workaround is to make it explicit that the poked value is a long, just add an **L**
```poke(c.io.PNewInst, 0xffffffffL)```

There's no easy way to fix this.  In Scala, the **0xffffffff** is converted to minus -1 before it is promoted to a BigInt as minus one, which has a very different value.  I have added a note about this  to the chisel testers [FAQ](https://github.com/freechipsproject/chisel-testers/wiki/Frequently-Asked-Questions).

oharboe, 2017-06-27:
What's weird is that it works in Verilator and fails in firrtl. I would
prefer that it fails in both.

chick, 2017-06-27:
That's a fair question, I'll look a little deeper

chick, 2018-03-07:
This is a bit of a hole but for the moment we are not going to change the behavior. Users using promotion of int to bigint implicitly will get this error with interpreter despite verilator behaving differently. The fix is simple, either specifying long or bigint explicitly and the fix works fine in both interpreter and verilator.

closed 2018-03-07 by chick