chipsalliance/chisel #636: Improve error message when forgetting to use cloneType

oharboe, 2017-06-29:
If you forget to use cloneType, then you'll get this error message which doesn't tell you what to do or where to do it:

*** RUN ABORTED ***
  chisel3.core.Binding$BindingException: : Already bound to WireBinding(ascenium.psm2.psm2@0)
  at chisel3.core.Binding$.AlreadyBoundException(Binding.scala:54)
  at chisel3.core.Element.binding_$eq(Bits.scala:33)
  at chisel3.core.Data$$anonfun$chiselCloneType$2.apply(Data.scala:266)
  at chisel3.core.Data$$anonfun$chiselCloneType$2.apply(Data.scala:265)
  at scala.collection.TraversableLike$WithFilter$$anonfun$foreach$1.apply(TraversableLike.scala:733)
  at scala.collection.Iterator$class.foreach(Iterator.scala:893)
  at scala.collection.AbstractIterator.foreach(Iterator.scala:1336)
  at scala.collection.IterableLike$class.foreach(IterableLike.scala:72)
  at scala.collection.AbstractIterable.foreach(Iterable.scala:54)
  at scala.collection.TraversableLike$WithFilter.foreach(TraversableLike.scala:732)
  ...

ducky64, 2017-06-29:
Is this still an issue in the latest chisel? #617 should give better error messages (along the line of "must be a chisel type, not hardware"), though it doesn't yet offer an resolution.

#578 also proposes to replace cloneType and friends with getType, which should be more intuitive.

oharboe, 2017-06-29:
this is w the may 30. chisel 3 snapshot.

ducky64, 2017-07-28:
Yeah, the PR that gives a better error was only merged June 26, so see if the errors are better now.
New APIs like chiselTypeOf are still under discussion.

closed 2018-03-05 by ducky64