chipsalliance/chisel #639: Build API for memory initialization from file.

chick, 2017-06-30:
Verilog supports the ```$readmem``` family of functions that can be used to load memory from a text or binary file.  We would like to allow some sort of annotation to memories that could specify a file to be read.  Considerations:
- Can this be done as a library, without having to build special case support into firrtl
- Can this file name be soft or does it have to be a fixed name and the external toolchain could move different files into place with this name in order to test different memory contents without having to re-build the DUT
- Can we use [Verilog Bind](//www.asic-world.com/systemverilog/assertions22.html) to accomplish this.
- Can the interpreter also recognize this same annotation to load memory

chick, 2017-06-30:
By doctoring up the Firrtl Verilog emitter it was easy enough to verify that the readmem can be used.
```scala
      def initialize_mem(s: DefMemory) {
        val index = wref("initvar", s.dataType)
        val rstring = rand_string(s.dataType)

//        initials += Seq("`ifdef RANDOMIZE_MEM_INIT")
//        initials += Seq("for (initvar = 0; initvar < ", s.depth, "; initvar = initvar+1)")
//        initials += Seq(tab, WSubAccess(wref(s.name, s.dataType), index, s.dataType, FEMALE),
//                             " = ", rstring,";")
//        initials += Seq("`endif // RANDOMIZE_MEM_INIT")

        initials += Seq("""$display("reading file\n") ;""")
        initials += Seq("""$readmemh("/Volumes/UCB-BAR/chisel-testers/data.hex", memory) ;""")
        initials += Seq("""$display("reading done\n") ;""")
      }
```
And a simple test, like 
```
class MemoryWrapper(val memorySize: Int) extends Module {
  val io = IO(new Bundle {
    val address = Input(UInt(8.W))
    val data    = Output(UInt(8.W))
  })

  val memory = Mem(memorySize, UInt(8.W))

  io.data := memory(io.address)
}

class MemoryWrapperTester(c: MemoryWrapper) extends PeekPokeTester(c) {
  for(addr <- 0 until c.memorySize) {
    poke(c.io.address, addr)
    step(1)
    step(1)
    println(f"mem[$addr%3d] is ${peek(c.io.data)}%8X")

  }
}
```
confirmed that memory can be loaded

ccelio, 2017-06-30:
Here's an example project that used `$readmemh` via a Verilog blackbox. Part of the challenge was this was used to load in the test binaries, so it wasn't a single hardcoded hex file.

https://github.com/librecores/riscv-sodor/blob/83eda919a4e0076ac0d3b9e1de298a4ff9328253/vsrc/AsyncReadMem.sv
https://github.com/librecores/riscv-sodor/blob/83eda919a4e0076ac0d3b9e1de298a4ff9328253/emulator/rv32_1stage/emulator.cpp#L72
https://github.com/librecores/riscv-sodor/blob/83eda919a4e0076ac0d3b9e1de298a4ff9328253/src/common/memory.scala#L71

chick, 2018-11-21:
Completed with PR #840

closed 2018-11-21 by chick