chipsalliance/chisel #640: For this code the time from firrtl to verilg could be very slow

apolonic, 2017-07-03:
I create a code (pasted following) which the run time from firrtl to verilog will be hours long. The running log looks like:

[mkdir: cannot create directory `target': File exists
warning: there was one deprecation warning; re-run with -deprecation for details
warning: there were 49 feature warnings; re-run with -feature for details
two warnings found
Run starting. Expected test count is: 2
Test:
[info] [0.002] Elaborating design...
[info] [0.712] Done elaborating.
- elaborate
[info] [0.000] Elaborating design...
[info] [0.255] Done elaborating.](url)
From this point the time will be very long to the end.

The scala compiler and code runner version 2.11.8

The code(something went wrong and the file can not be processed so I just pasted here):
[import chisel3._
import chisel3.util._

object ce_pm{
  val div = 4
  val e = 1   
  val ec= 1   
  val p = 10 // 16/div        
  val s = p*p       
  val w = s*e       

  val ext = 64      
  val extw= ext*e   

  val irp = 10 // 40/div  // InREG parameter
  val irn = irp*irp // InREG reg number
}

class Mux4(n: Int) extends Module {
  val io = IO(new Bundle{
    val i = Input(Vec(4,UInt(n.W)))
    val s = Input(UInt(2.W))
    val o = Output(UInt(n.W))
  })
  val mux00 = Wire(UInt(n.W))
  val mux01 = Wire(UInt(n.W))
  mux00 := Mux(io.s(0)===1.U,io.i(1),io.i(0))
  mux01 := Mux(io.s(0)===1.U,io.i(3),io.i(2))
  io.o  := Mux(io.s(1)===1.U,mux01,mux00)
}

class CEIO_TwoD_Torus extends Bundle {
  val n = Input(UInt(ce_pm.e.W))
  val s = Input(UInt(ce_pm.e.W))
  val w = Input(UInt(ce_pm.e.W))
  val e = Input(UInt(ce_pm.e.W))
}

class TwoD_TorusReg extends Module {
  val io = IO(new Bundle{
    val i = new CEIO_TwoD_Torus()
    val o = new CEIO_TwoD_Torus().flip
    val d = Input(UInt(ce_pm.e.W)) // set data
    val c = Input(Vec(4,UInt(1.W)))
    // bit0,en;bit2:1,shift;bit3:set
  })
  val r = Reg(UInt(ce_pm.e.W),init=0.U)
  val u_mux4 = Module(new Mux4(ce_pm.e))
  u_mux4.io.i(0) := io.i.e
  u_mux4.io.i(1) := io.i.s
  u_mux4.io.i(2) := io.i.w
  u_mux4.io.i(3) := io.i.n
  u_mux4.io.s    := Cat(io.c(2),io.c(1))
  when (io.c(0) === 1.U) {
    when (io.c(3) === 0.U) {
      r := u_mux4.io.o
    } .otherwise {
      r := io.d
    }
  } 
  io.o.e := r
  io.o.s := r
  io.o.w := r
  io.o.n := r
}

class Top extends Module {
  //5 bit addr to write to REG(2 bit shift when we=0),1 bit we, 1 bit en
  val io = IO(new Bundle{
    val i = Input (UInt(ce_pm.extw.W))
    val o = Output(Vec(ce_pm.p,Vec(ce_pm.p,UInt(ce_pm.e.W))))
    val c = Input (UInt(7.W))
  })
  val n  = ce_pm.irp
  val r_vec = (0 to n-1).map ( i=>
                (0 to n-1).map ( j=>
                  Module(new TwoD_TorusReg)
                )
              )
  for (i <- 0 to n-1) {
    for (j <- 0 to n-1) {
      //r_vec(i)(j).io.c := Cat(io.c(1),io.c(3),io.c(2),io.c(0)).asBits
      r_vec(i)(j).io.c(0) := io.c(1)
      r_vec(i)(j).io.c(3) := io.c(0)
      r_vec(i)(j).io.c(2) := io.c(2)
      r_vec(i)(j).io.c(1) := io.c(3)
    }
  }
  // out
  val m = ce_pm.p
  (0 to m-1).map(i=>{
    (0 to m-1).map(j=>{
      io.o(i)(j) := r_vec(i)(j).io.o.e
    })
  })
  //2-D-Torus interconnection
  (1 to n-1).map(i=>{
    (1 to n-1).map(j=>{
      r_vec(i)(j).io.i.w := r_vec(i)(j-1).io.o.e
      r_vec(i)(j).io.i.n := r_vec(i-1)(j).io.o.s
    })
  })
  (0 to n-2).map(i=>{
    (0 to n-2).map(j=>{
      r_vec(i)(j).io.i.e := r_vec(i)(j+1).io.o.w
      r_vec(i)(j).io.i.s := r_vec(i+1)(j).io.o.n
    })
  })
  (0 to n-1).map(i=>{
    r_vec(i)(0).io.i.w := r_vec(i)(n-1).io.o.e
    r_vec(0)(i).io.i.n := r_vec(n-1)(i).io.o.s
  })
}](url)

chick, 2017-07-03:
In the 2-D-Torus interconnection blocks you should be using foreach instead of map. The map will be creating a lot of collections that are just thrown away.
> This is not the source of your long compilation times, just a note on Scala/Chisel best practices.

apolonic, 2017-07-04:
@chick  thanks for the advance, I also like to try but shame on myself that I can't find a example of using foreach on a 2D Vec or Array, I tried something like this but syntex is failed
    val n_vec = (0 to 2).map(i=>
                  (0 to 2).map(j=>
                    Wire(UInt(3.W))
                  )
                ) 
    n_vec.foreach((i:Int)=>(
      i.foreach((j:Int)=>{
        n_vec(i)(j) := i + j
      })
    ))

apolonic, 2017-07-04:
@chick  Could you enlight me how to use `foreach` for the 2d list? I update `map` to `for` body. Thanks for the advance, it can be faster a little (when ce_pm.irp=10, elabrate time decrease from 0.28 to 0.23). You're right there is still other source behind (when ce_pm.irp=20, time costs still too long)
update the class `Top` in for body
```
class Top extends Module {
  val io = IO(new Bundle{
    val i = Input (UInt(ce_pm.extw.W))
    val o = Output(Vec(ce_pm.p,Vec(ce_pm.p,UInt(ce_pm.e.W))))
    val c = Input (UInt(7.W))
  })
  val n  = ce_pm.irp
  val r_vec = (0 to n-1).map ( i=>
                (0 to n-1).map ( j=>
                  Module(new TwoD_TorusReg)
                )
              )
  for (i <- 0 to n-1) {
    for (j <- 0 to n-1) {
      r_vec(i)(j).io.c(0) := io.c(1)
      r_vec(i)(j).io.c(3) := io.c(0)
      r_vec(i)(j).io.c(2) := io.c(2)
      r_vec(i)(j).io.c(1) := io.c(3)
    }
  }
  // out
  val m = ce_pm.p
  for (i <- 0 to m-1) {
    for (j <- 0 to m-1) {
      io.o(i)(j) := r_vec(i)(j).io.o.e
    }
  }
  //2-D-Torus interconnection
  for (i <- 1 to n-1) {
    for (j <- 1 to n-1) {
      r_vec(i)(j).io.i.w := r_vec(i)(j-1).io.o.e
      r_vec(i)(j).io.i.n := r_vec(i-1)(j).io.o.s
    }
  }
  for (i <- 0 to n-2) {
    for (j <- 0 to n-2) {
      r_vec(i)(j).io.i.e := r_vec(i)(j+1).io.o.w
      r_vec(i)(j).io.i.s := r_vec(i+1)(j).io.o.n
    }
  }
  for (i <- 0 to n-1) {
    r_vec(i)(0).io.i.w := r_vec(i)(n-1).io.o.e
    r_vec(0)(i).io.i.n := r_vec(n-1)(i).io.o.s
  }
}
```

chick, 2017-07-04:
@apolonic If you are referring to the ```map``` statements here
```scala
  val r_vec = (0 to n-1).map ( i=>
                (0 to n-1).map ( j=>
                  Module(new TwoD_TorusReg)
                )
              )
```
You cannot use foreach in this case, so ```map``` is ok, but the simplest way of getting your 2d data structure is
```scala
  val r_vec = Array.fill(n, n)(Module(new TwoD_TorusReg))
```
Checkout [Creating collections from scratch](http://docs.scala-lang.org/overviews/collections/creating-collections-from-scratch.html) for a handy cheatsheet.  I like to use tabulate when the elements  I am creating depend on their indices.  In the case at hand though, fill is the proper way.

apolonic, 2017-07-04:
@chick  Hi, did you also can see the *.fir is generated while the whole processing is running? BTW, I did some test the response/log you can see here https://stackoverflow.com/questions/44849211/can-chisel-translates-firrtl-to-verilog-in-parallel-multi-cpu/44901620#44901620

ducky64, 2017-11-08:
This has been inactive for a while, if it's still an issue feel free to reopen.

closed 2017-11-08 by ducky64