chipsalliance/chisel #642: Stack overflow problem for small ROM

oharboe, 2017-07-05:
I've implemented a ROM as proposed by https://github.com/freechipsproject/chisel3/wiki/Memories but when I try to run the tests, it stack-overflows:
`[[35minfo[0m] [0.003] Elaborating design...
[[35minfo[0m] [0.202] Done elaborating.
[[35minfo[0m] [0.000] Elaborating design...
[[35minfo[0m] [0.078] Done elaborating.
End of dependency graph
Circuit state created
*** RUN ABORTED ***
  java.lang.StackOverflowError:
  at firrtl_interpreter.LoFirrtlExpressionEvaluator.evaluate(LoFirrtlExpressionEvaluator.scala:263)
  at firrtl_interpreter.LoFirrtlExpressionEvaluator.binaryBitWise(LoFirrtlExpressionEvaluator.scala:241)
  at firrtl_interpreter.LoFirrtlExpressionEvaluator.evaluate(LoFirrtlExpressionEvaluator.scala:355)
  at firrtl_interpreter.LoFirrtlExpressionEvaluator.evaluate(LoFirrtlExpressionEvaluator.scala:290)
  at firrtl_interpreter.LoFirrtlExpressionEvaluator$$anonfun$5.apply(LoFirrtlExpressionEvaluator.scala:453)
  at firrtl_interpreter.LoFirrtlExpressionEvaluator$$anonfun$5.apply(LoFirrtlExpressionEvaluator.scala:445)
  at firrtl_interpreter.Timer.apply(Timer.scala:40)
  at firrtl_interpreter.LoFirrtlExpressionEvaluator.firrtl_interpreter$LoFirrtlExpressionEvaluator$$resolveDependency(LoFirrtlExpressionEvaluator.scala:444)
  at firrtl_interpreter.LoFirrtlExpressionEvaluator.getValue(LoFirrtlExpressionEvaluator.scala:81)
  at firrtl_interpreter.LoFirrtlExpressionEvaluator.evaluate(LoFirrtlExpressionEvaluator.scala:304)
  ...
`

oharboe, 2017-07-05:
[console.zip](https://github.com/freechipsproject/chisel3/files/1124560/console.zip)

chick, 2017-07-05:
I cannot reproduce the problem.  This runs for me with both the *firrtl* and *verilator* backends.  The only changes I made were
1. Comment out ```//import util.control.Breaks._```
1. Change the third expect to ```expect(c.io.character, c.helloWorldString(i % l).toInt)```

I tried fiddling with memory parameters, the following:
```
export SBT_OPTS='-Xmx64M -XX:+UseConcMarkSweepGC -XX:+CMSClassUnloadingEnabled -Xss2M  -Duser.timezone=GMT'
```
could cause some error. But didn't seem to be related to yours.  I usually use -Xmx2G in SBT_OPTS

@jackkoenig can you answer the strange verilog question

oharboe, 2017-07-06:
>    2. Change the third expect to expect(c.io.character,

When it fails depends on the JVM settings as you point out, so that would
explain why I would see it and you wouldn't.

The length of "val helloWorldString" is what seems to trigger it.

Can you try to make it e.g. 64kBytes?

Perhaps the recursive algorithm is explained by the funny Verilog it
produces?

jackkoenig, 2017-07-06:
@oharboe Per the "strange" Verilog: synthesis tools do in fact infer this correctly. We emit Verilog like this because it's easier to implement and still handled well by existing CAD tools. We would like to emit more traditional Verilog vectors but there are many technical hurdles to overcome before then so it will be a while.

@chick Long dependency chains could definitely cause a stack overflow error. We should make evaluate algorithm tail recursive or use an explicit stack.

jackkoenig, 2017-10-19:
Is this still a problem?

oharboe, 2017-10-20:
Yes.

Updated test files here: [console2.zip](https://github.com/freechipsproject/chisel3/files/1400875/console2.zip)

`[[35minfo[0m] [0.002] Elaborating design...
[[35minfo[0m] [0.138] Done elaborating.
Total FIRRTL Compile Time: 1409.1 ms
Total FIRRTL Compile Time: 256.3 ms
End of dependency graph
Circuit state created
*** RUN ABORTED ***
  java.lang.StackOverflowError:
  at scala.math.BigInt$.int2bigInt(BigInt.scala:97)
  at firrtl_interpreter.ConcreteUInt.<init>(Concrete.scala:332)
  at firrtl_interpreter.Concrete$.apply(Concrete.scala:292)
  at firrtl_interpreter.LoFirrtlExpressionEvaluator.evaluate(LoFirrtlExpressionEvaluator.scala:374)
  at firrtl_interpreter.LoFirrtlExpressionEvaluator.comparisonOp(LoFirrtlExpressionEvaluator.scala:179)
  at firrtl_interpreter.LoFirrtlExpressionEvaluator.evaluate(LoFirrtlExpressionEvaluator.scala:332)
  at firrtl_interpreter.LoFirrtlExpressionEvaluator.binaryBitWise(LoFirrtlExpressionEvaluator.scala:242)
  at firrtl_interpreter.LoFirrtlExpressionEvaluator.evaluate(LoFirrtlExpressionEvaluator.scala:355)
  at firrtl_interpreter.LoFirrtlExpressionEvaluator.evaluate(LoFirrtlExpressionEvaluator.scala:290)
  at firrtl_interpreter.LoFirrtlExpressionEvaluator$$anonfun$5.apply(LoFirrtlExpressionEvaluator.scala:453)
  ...
`

chick, 2017-10-20:
I am not sure how I missed this one.  It's a bit strange.
I get the same stack overflow if I run this from chisel, but if I take either the fir or lo.fir file I can read it into the interpreter repl without a problem.  I'll try and figure it out tomorrow

oharboe, 2017-10-20:
Would it make sense to add this code to the automated regression test suite
in git?

chick, 2017-10-20:
This appears to be a straight up legitimate stack overflow.  The interpreter is not elegant in it's dependency resolution, the dependency depth here was just too big for it.  I was able to run using
```
export SBT_OPTS="-Xmx2G -XX:+UseConcMarkSweepGC -XX:+CMSClassUnloadingEnabled -XX:MaxPermSize=2G -Xss2M  -Duser.timezone=GMT"

sbt `testOnly examples.ConsoleTester`
``` 
I think this is a fair amount of work to fix properly, but alternatively I am in development of a next generation interpreter that has a fairly different (and hopefully quite a bit faster) execution engine that will not have this kind of dependency depth.
When that is done (pretty soon) we should include this in the regression.

oharboe, 2017-10-20:
makes sense. thanks!

chick, 2017-11-06:
fast-interpreter is still  a ways away, but I have incorporated this test into it's test suite and it passes with new scheme.

closed 2017-11-06 by chick