chipsalliance/chisel #646: chiselName should be able to name signals in anonymous class instances

colinschmidt, 2017-07-14:
I have some code that looks like:
```
val scope = new {
  val coolWire = io.blah && io.foo
}
```
It would be great if I could annotate this scope with chiselName and get the wire name as
```
scope_coolWire
```
Currently it tells me I can only use the annotation on functions and modules.

jackkoenig, 2017-07-14:
I think the way this would have to work is chiselName would rewrite the anonymous class into a "named" class that gets the annotation and then instantiates it. Should be doable I think though

@ducky64 @sdtwigg

ducky64, 2017-07-14:
This should be pretty straightforward to do, and there are several options for implementation paths.

I think the more interesting question is the interface to present. Are you just creating the inner class / object just to give it a scope, or is there some other reason it's done? Is the `new` also needed?

Related: what about iterated loops and variable interpolation in general (like splicing loop indices into generated names)?

sdtwigg, 2017-07-14:
This is different than what I thought you meant when you asked about annotating anonymous classes.

In this case, you can piggyback on using the annotation on the enclosing class to look into the parts of the anonymous class. Also, it seems like (at the very least) having the name "coolWire" would be preferable. (Although, you would also see the enclosing val name during the traversal, this might be tricky to pick up properly)

As an aside though, why do you have code like that...?

colinschmidt, 2017-07-17:
The code looks like this:
```
val scope = new {
  val coolWire = blah
  def coolFunc(n: UInt) = blah && coolWire
}
...
scope.coolFunc(n)
...
scope.coolFunc(n)
...
```
I end up having a few "scopes" with the same "funcs" so this gives me a nice way to avoid naming collisions as well as a way to write certain parts of the logic in a certain domain like the following:

``` {import scope._; func1(x); func2(y); func3(z)}```

This is mostly in the name of self-documenting code and brevity. I could obviously name my functions and wires with a "scope" prefix but it seems like a nice language feature that would be not too difficult to support naming.

P.S. I didn't come up with this coding style but it seems pretty neat and the end product is a nicely compartmentalized readable description of the logic.

jackkoenig, 2022-03-22:
Fixed via the naming plugin in 3.4.

closed 2022-03-22 by jackkoenig