chipsalliance/chisel #649: Chisel is not inferring the correct width of register

lorenzoditucci, 2017-07-20:
Hi, 

I have experienced some issues in trying to have Chisel inferring the correct number of widths for a register.

In particular, the register should be 6-bits, but instead, a 1-bit register is defined in the Verilog.

I tried to create an example code to reproduce the issue:

```
class width_error extends Module{
  val io = IO(new Bundle{
    val p1 = new HMC_port()
  })

  val regTag = Reg(init = 0.U)
  val readCounter = Counter(100)

  when(readCounter.value < 50.U){
    io.p1.cmd.bits.tag := regTag
    regTag := regTag + 1.U
    readCounter.inc()
  }

}
```

where HMC_port is defined as follows

```
class HMC_port extends Bundle{

  val cmd = Decoupled(new HMCCommand)
  val wr = DecoupledIO(UInt(128.W))
  val rd = Decoupled(new HMCResponseChannel).flip //READY IS NOT USED!!

  override def cloneType = {new HMC_port().asInstanceOf[this.type ]}
}

class HMCCommand extends Bundle{
  val addr = UInt(34.W)
  val size = UInt(4.W)
  val tag = UInt(6.W)
  val cmd = UInt(4.W)

  override def cloneType = { new HMCCommand().asInstanceOf[this.type] }
}

class HMCResponseChannel extends Bundle{
  val data = UInt(128.W)
  val errstat = UInt(7.W)
  val dinv = UInt(1.W)
  val tag = UInt(6.W)

  override def cloneType = { new HMCResponseChannel().asInstanceOf[this.type ]}
}
```
I attach also a version of the Verilog that is produced from this code.

[width_error.v.zip](https://github.com/freechipsproject/chisel3/files/1163621/width_error.v.zip)

chick, 2017-07-20:
I assume you are talking about the width of regTag.  regTag is taking the width from the init statement.  The width of 0.U is one.  When creating registers with an initial value you should specify the desired width as part of the initializing constant, in your case, the proper line should be something like
```
val regTag = RegInit(0.U(6.W)
```

ccelio, 2017-07-20:
Chisel does **not** back-propogate width inference, only forward-propogation up to the specified IO `output`s.

```scala
  val regTag = Reg(init = 0.U) // creates a register of unknown width
  ...
  regTag := regTag + 1.U // adds a one-bit wire to a register of unknown width
  ...
  // chisel is NOT going to back-propogate width-inference, so a 1-bit register is written into a 6-bit output.
  io.p1.cmd.bits.tag := regTag 
```

ccelio, 2017-07-20:
Thanks @lorenzoditucci, I have opened an issue #650  to help us track adding the proper documentation for this behavior.

closed 2017-07-20 by ccelio

sdtwigg, 2017-07-20:
As an addendum:

val regTag = Reg(init = 0.U)
regTag := regTag + 1.U

The init adds the constraint that width >= 1
The + is a non-extending add so the connection adds the constraint that width >= max(width, 1)

The width inference attempts to minimize widths so the result is width=1. Note that using +& would result in an error (assuming there are no remaining FIRRTL bugs surrounding this issue).

lorenzoditucci, 2017-07-20:
ok I see, thanks for the help,  I did not know that. 

I was also wondering if would not be worthy to at least generate a Warning for such a behavior.

sbeamer, 2017-07-20:
Following up on the point made by @lorenzoditucci, how might a warning have made this less unintuitive?

Could this particular case (register declared without width and width of input to it is unknown) be detected?

Additionally, the IO assignment `io.p1.cmd.bits.tag := regTag` leaving 5 bits unassigned is also troubling. I don't want to reignite the debate about warnings for unconnected IOs, but perhaps in this special case where the unconnected IOs are due to width inference, this could be also be detected.

Either way, +1 for clearing this up in the docs.