chipsalliance/chisel #650: Documentation needed on how Chisel/FIRRTL width-inference works

ccelio, 2017-07-20:
Related: #649

The fact that Chisel (or perhaps rather FIRRTL) width inference is forward propagating needs to be documented.

For example: 
* The Chisel [FAQ](https://github.com/freechipsproject/chisel3/wiki/Frequently-Asked-Questions)
* [A short user guide to chisel](https://github.com/freechipsproject/chisel3/wiki/Short-Users-Guide-to-Chisel)

The only documentation I found is [Getting Started](https://chisel.eecs.berkeley.edu/2.2.0/getting-started.html) from 2015, but is a bit buried.

jackkoenig, 2017-08-04:
I think is super important to 3.0.0 since it is a key part of the API that needs to be formally described and tested.

I brought this up a while ago (https://github.com/freechipsproject/chisel3/pull/455#discussion_r101434653) but it got put on the back burner a bit.

chick, 2017-08-22:
I updated the section on width inference in the chisel3 wiki, last week.  I also referred reader to the firrtl spec, which should probably be the final formal arbiter on how this works

closed 2017-08-22 by chick