chipsalliance/chisel #652: On the deprecation of log2Up

da-steve101, 2017-07-23:
1) I don't think log2Up should reference log2Ceil in the deprecation warning. log2Ceil( 0 ) = 0 but log2Up( 0 ) = 1 so slightly different functionality.
2) As these are functionally different, I am in favour of keeping log2Up within chisel3

ducky64, 2017-07-25:
See #528 for a rationale of why, log2Up was originally introduced to work around the lack of zero width wires, a feature which now exists and should present a cleaner solution.

Is there a particular reason (not addressed above) that you want to keep using log2Up for?

da-steve101, 2017-07-27:
It is related to that i guess. I can see the point of deprecating it. I was just frustrated when I changed my code log2Ceil and then some of my stuff broke when i was using:
```scala
val xVal = 0
val x = xVal.U( log2Up( xVal ).W )
val x = xVal.U( log2Ceil( xVal ).W )
```

jackkoenig, 2017-07-27:
Do you need to declare the width or would your code work just fine letting
it be inferred since it's a literal?

Also is putting 0 as the width failing? Supporting 0 as the width is part
of supporting of zero-width wires.

da-steve101, 2017-07-29:
no zero width is fine, but declaring 0 with a width of zero results in "insufficient bits" error. Yeah I could. It was chisel2 code and i just did a find and replace.

jackkoenig, 2017-08-02:
So are you still arguing for keeping log2Up or should I close this issue?

closed 2017-08-04 by jackkoenig

jkorinth, 2017-08-25:
Still, currently `log2Ceil(1) == 0`, which seems like a bug? It broke some of my bundles, e.g., when declaring a 1-bit input as `val x = Input(UInt(log2Ceil(someValue).W))` and having `someValue == 1` makes `x` disappear from the bundle completely (leading to further errors in `PeekPokeTester` etc.).