chipsalliance/chisel #656: Proposal: Support creation of artifacts by chisel toolchain

chick, 2017-07-31:
There are several situations where firrtl passes may create verilog or other files during execution.  Code higher in the execution hierarchy needs to access these files but a good system for communicating this information upwards does not exists.  
Use cases:

- Black Box verilog files are handled by naming convention currently, discovering these files and referencing them in VCS and verilator toolchains is done by convention and is ugly
- Proposed method of memory initialization would use verilog binding to augment chisel produced verilog.  These files would be created in firrtl transformation and need to be known by testers
- Proposed re-introduction of simple dot graphics would create dot files in firrtl transformation, needs to be accessible by higher toolchain elements, jupyter notebook would be a great use case

colinschmidt, 2017-08-01:
I'm confused about the name/description of this request. You say firrtl may create files that chisel needs but the request if for chisel to create things.
Is the idea to drive the firrtl creation of files from chisel with something like annotations?
Or is the idea to just formalize an API for creating artifacts from chisel so users like rocket-chip don't roll their [own](https://github.com/freechipsproject/rocket-chip/blob/6ef8ee5d4d1694e0e465720f0cd8ba14609d8521/src/main/scala/util/GeneratorUtils.scala#L125)?

azidar, 2018-03-09:
Solved with new annotations. Will add CircuitState to FirrtlExecutionResult, so they are more easily accessible.

closed 2018-03-09 by azidar