chipsalliance/chisel #657: Reg binding problems with Rocket

grebe, 2017-08-01:
I'm playing around with the APB stuff in rocket chip and it seems to be leading to some binding problems.

Here is the relevant portion of the stacktrace:
```
chisel3.core.Binding$RebindingException: Attempted reassignment of binding to HeterogeneousBag(freechips.rocketchip.amba.apb.APBBundle@377)
	at chisel3.core.Data.binding_$eq(Data.scala:221)
	at chisel3.core.Aggregate.bind(Aggregate.scala:19)
	at chisel3.core.Aggregate$$anonfun$bind$1.apply(Aggregate.scala:23)
	at chisel3.core.Aggregate$$anonfun$bind$1.apply(Aggregate.scala:22)
	at scala.collection.Iterator$class.foreach(Iterator.scala:742)
	at scala.collection.AbstractIterator.foreach(Iterator.scala:1194)
	at scala.collection.IterableLike$class.foreach(IterableLike.scala:72)
	at scala.collection.AbstractIterable.foreach(Iterable.scala:54)
	at chisel3.core.Aggregate.bind(Aggregate.scala:22)
	at chisel3.core.BaseModule.IO(Module.scala:249)
	at Chisel.package$CompatibilityModule._autoWrapPorts(compatibility.scala:242)
	at chisel3.core.requireIsHardware$.apply(Binding.scala:26)
	at chisel3.core.RegInit$.apply(Reg.scala:92)
	at chisel3.core.RegInit$.apply(Reg.scala:78)
	at freechips.rocketchip.amba.axi4stream.APBInStreamOutFIFOModule.<init>(MMAPFIFO.scala:120)
```

MMAPFIFO.scala:120 is a `RegInit(0.U(64.W))` inside a `LazyModuleImp` that has a `APBRegisterRouter` ([see here](https://github.com/freechipsproject/rocket-chip/blob/4c595d175c2c3d3dfa721a78a4a2a9fab009621b/src/main/scala/amba/apb/RegisterRouter.scala)). The error seems to be happening when auto-wrapping an `APBBundle` (see [here](https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/amba/apb/Bundles.scala)) that was originally associated with the register router.

Switching to the compatibility layer didn't seem to help. Am I doing something dumb? I don't really understand how all this binding stuff is meant to work with rocket.

ducky64, 2017-08-01:
Is there a link to MMAPFIFO.scala:120, I cant seem to find it in the rocket-chip repo.

A rebinding error basically means you tried to use a hardware type (for example, Reg) like a data type (for example, UInt), which are now more strictly separated. So something like Reg(Reg(UInt())) is not fine.

grebe, 2017-08-01:
MMAPFIFO is something I'm writing. 

I think I tracked it down. I thought this might have been some strange interaction between chisel3 and the compatibility layer, but I don't think that's the case. [The code used to get the bundle type](https://github.com/freechipsproject/rocket-chip/blob/573890e102471bee1e0b60e0dec48fa5c2a7e853/src/main/scala/diplomacy/Nodes.scala#L229) wraps it in a wire for some reason, which is probably not a good thing to do. For now, I'm just calling cloneType on this wire, but the right thing to do is probably to get rid of that call to `Wire()`. In any case, I don't think this is a chisel problem.

closed 2017-08-01 by grebe