chipsalliance/chisel #661: How to do such a supper 'bulk connection'?

apolonic, 2017-08-03:
the bulk connection to me is like a great function which can search for the pair interfaces who have the same name and connect them, but the pairs have to be complete for both sides of the bulk. What I want is to search and connect the pairs only within the left side, eg.

class ax extends Bundle {
  val a0 = Input(Bool())
  val a1 = ...
}
class bx extends Bundle {
  val b0 = Input(Bool())
  val b1 = ...
}
class cx extends Bundle {
  val a0 = Input(Bool())
  val a1 = ...
  val b0 = Input(Bool())
  val b1 = ...
}
class A extends Module {
  val io = IO(new Bundle{
    val a = new ax()
  })
}
class B extends Module {
  val io = IO(new Bundle{
    val b = new bx()
  })
}
class C extends Module {
  val io = IO(new Bundle{
    val c = new cx
  })
  val ua = new A
  val ub = new B
  // this is what I want to do
  ua.io.a <> io.c
  ub.io.b <> io.c
}

colinschmidt, 2017-08-03:
Hmm not just I understand your question but why not just do something like
```
class cx extends Bundle {
val a = new ax()
val b = new bx()
}
...
ua.io.a <> io.c.a
ub.io.b <> io.c.b
```

apolonic, 2017-08-04:
It brings the name of io_a_a0 and io_c_a_a0.
The main cause is to integrate other guy's verilog RTL as an class extended of BlackBox. Integration of the top level involves modules of IP and such things, since I can't regular every interface to this naming rule, either I create one more hierarchy to wrap the IP into a new module, or I can use this "just left side bulk connections" function to resolve it.

apolonic, 2017-08-04:
Is there a way to 'virtually' bund more bundles together to a new 'virtual bundle' and then do bulk connetion?

apolonic, 2017-08-04:
Is there a way to traversal all signals in the bundle and do map or connection?

jackkoenig, 2017-10-19:
It looks to me like @apolonic actually wants Chisel2 bulk connection semantics or kind of an "unsafe bulk connect". This could possibly be resolved via a "setName" API or via clever inheritance and "connectAsParent"

edwardcwang, 2018-11-01:
A similar usecase is: two bundles share a common trait which defines some I/Os but there's no way in pure chisel3 semantics to connect the shared trait together.

tampler, 2018-11-17:
I have a solution with abstract types. Let me know if this works for you. If not, I got another idea. 

```scala
abstract trait Common extends Bundle {
  val c0 = Bool()
  val c1 = Bool()

  // Redefine this in derived classes
  type Bus 
}

trait Foo extends Common
trait Bar extends Common

class A extends Bundle with Foo {
  // doesn't work for = UInt (10.W) thou
  type Bus = UInt
}

class B extends Bundle with Bar /* with Foo with Bar with ... */ {
  type bb = UInt
}

// A BlackBox that has the following I/Os
// c0, c1, aa
class MyBBoxA extends BlackBox {
 val io = IO(Input(new A))
}

// A BlackBox that has the following I/Os
// c0, c1, bb
class MyBBoxB extends BlackBox {
 val io = IO(Output(new B))
}

class Mro extends Module {
 val io = IO(new Bundle {})
 val mod1 = Module(new MyBBoxA)
 val mod2 = Module(new MyBBoxB)

 mod1.io <> mod2.io // <-- now compiles

 // Non-DRY code, want to avoid
 mod1.io.c0 <> mod2.io.c0
 mod1.io.c1 <> mod2.io.c1
}

```

edwardcwang, 2018-11-18:
Thanks Boris for looking at this. Unfortunately, this doesn't work for as Chisel only recognizes instances of Chisel types as fields of a Bundle e.g. `val x = UInt(10.W)`, not the types themselves e.g. `val bad = UInt`. We can't use the Chisel types themselves `type t = ...` as fields for a Bundle, so the proposed solution doesn't appear to solve this problem.

tampler, 2018-11-20:
@edwardcwang I was actually able to generate both FIR and Verilog outputs with connected instances  as below: 
FIR:
    mod1.c1 <= mod2.c1 @[Mro.scala 54:10]
    mod1.c0 <= mod2.c0 @[Mro.scala 54:10]
Verilog:
  assign mod1_c1 = mod2_c1; // @[Mro.scala 54:10]
  assign mod1_c0 = mod2_c0; // @[Mro.scala 54:10]

If this doesn't work for you, pls upload your full example so that we could see how to solve your particular issue

edwardcwang, 2018-11-20:
The problem is that the bundle `A` is missing `aa` as a field and bundle `B` is missing `bb` as a field. As I wrote above, `val aa = UInt` does NOT declare a field, and using the proper declaration of `val aa = UInt(10.W)` results in the old bulk connect error since the members of both bundles are not the same.

```verilog
  MyBBoxA mod1 (
    .c1(mod1_c1),
    .c0(mod1_c0)
    // missing .aa(...)
  );
  MyBBoxB mod2 (
    .c1(mod2_c1),
    .c0(mod2_c0)
    // missing .bb(...)
  );
```

See the following snippet of Chisel:

```scala
abstract trait Common extends Bundle {
  val c0 = Bool()
  val c1 = Bool()
  // Redefine this in derived classes
  type Bus 
}
trait Foo extends Common {
  val aa = UInt // <-- changing this to UInt(10.W) declares a proper field but we will then run into the problem described in this issue, just with an extra layer of indirection (Foo and Bar are unnecessary)
}
trait Bar extends Common {
  val bb = UInt
}
class A extends Bundle with Foo {
  type Bus = UInt
}
class B extends Bundle with Bar /* with Foo with Bar with ... */ {
  type Bus = UInt
}

// A BlackBox that has the following I/Os
// c0, c1, aa
class MyBBoxA extends BlackBox {
 val io = IO(Input(new A))
}

// A BlackBox that has the following I/Os
// c0, c1, bb
class MyBBoxB extends BlackBox {
 val io = IO(Output(new B))
}

class Mro extends Module {
 val io = IO(new Bundle {})
 val mod1 = Module(new MyBBoxA)
 val mod2 = Module(new MyBBoxB)

 mod1.io <> mod2.io
 // Non-DRY code, want to avoid
/*
 mod1.io.c0 <> mod2.io.c0
 mod1.io.c1 <> mod2.io.c1
 */
 mod1.io.aa := 5.U
 mod2.io.bb := 6.U
}
```

tampler, 2018-11-20:
@edwardcwang Okay, I see what causes the problem. 

Its part of the core logic. We need to decide whether it is acceptable to bulk connect two records with partially different fields. Currently, its prohibited and hardcoded here: https://github.com/freechipsproject/chisel3/blob/abee251c262c1d60487153b1227aa587dc39a5cf/chiselFrontend/src/main/scala/chisel3/core/BiConnect.scala#L157

It is rather possible to allow partial bulk connections. But that adds complexity to the implementation and especially test cases. It is also a potential source of subtle bugs in RTL.

I do have a workaround, but it's a hack for this problem, rather than a nice language feature. I'd close this issue and push users to plan their signals upfront if they want to scrap the boilerplate.

edwardcwang, 2018-11-20:
Yep, this is exactly what this issue is trying to resolve...

azidar, 2020-10-27:
Could be supported with new connection operators.

chrisbaldwin2, 2021-02-26:
Looks like someone beat me to the punch. Here is a better formatted version of my post

When using the Bundle Assignment, all elements must match between the two objects. What I would really enjoy is a wildcard assignment that allows for bundles to be split and joined, without needing to write out every signal. This helps a ton with managing a datapath where all of the signals can be wrapped in a bundle, then easily passed from one module to the next similar to the Verilog ```(.*)```. 
Additionally the ```.*``` assignment could be done by a syntax like so
```moduel_i.io <*> _```
which could assign the bundle to the variables in the current context.

Currently when assigning two bundles that don't match, you get the following

> [error] chisel3.internal.ChiselException: Connection between left (anon$2(IO io_ctrlDataOut in ControlFile)) and source (anon$2(IO in unelaborated CPUFile)) failed @.memWData: Left Record missing field (memWData).

I think that it would be better to have an explicit operator to prevent users from accidentally missing signals, but it would be nice if people wanted to "live on the edge" and use wildcards to shorten laborious assignments.

jackkoenig, 2022-01-20:
This isn't necessarily fixed in the way everyone wanted, but `.viewAsSuperType` and `DataView` (new in Chisel 3.5.0) more generally enables such connections in a much more elegant way than before. I'm closing this issue and suggest everyone look at the documentation: https://www.chisel-lang.org/chisel3/docs/cookbooks/dataview.html#how-do-i-connect-a-subset-of-bundle-fields.

closed 2022-01-20 by jackkoenig