chipsalliance/chisel #662: How to traversal or return the signalName of all elements of a bundle?

apolonic, 2017-08-04:
Could you enlight me how to traversal the signalName of a bundle?
like
class test extends Bundle {
  val io = IO(new Bundle{
    val a = Input(Bool())
    val b = Input(Bool())
  })
  for ( signame <- io.elements) {
    printf("%s \n",signame)
  }
}

jackkoenig, 2017-08-04:
The type of [elements](https://chisel.eecs.berkeley.edu/api/index.html#chisel3.core.Bundle@elements:scala.collection.immutable.ListMap[String,chisel3.core.Data]) is `ListMap[String, Data]`, so the correct way to iterate on the elements of a Bundle is:
```scala
for ((name, data) <- io.elements) {
  printf(s"$name -> %d\n", data)
}
```

apolonic, 2017-08-07:
class AIO extends Bundles {
  val a0 = Input(Bool())
}
class AM extends Module {
  val io = IO(new Bundle {
    val a = new A().flip
  })
}
class BM extends Module {
  val io = new A()
  val u0 = Module(new AM())
  for ((name,d) <- u0.io.a) {
    printf (s"$name <- %d\n",d)
  }
}

error: value filter is not a member of AIO
  for ((n,d) <- u0.io.a) {
                              ^
error: type mismatch;
 found   : Any
 required: chisel3.core.Bits
    printf(s"$name <- $d\n",d)
                                   ^

apolonic, 2017-08-07:
I got errors on the code:

class AIO extends Bundles {
  val a0 = Input(Bool())
}
class AM extends Module {
  val io = IO(new Bundle {
    val a = new A().flip
  })
}
class BM extends Module {
  val io = new A()
  val u0 = Module(new AM())
  for ((name,d) <- u0.io.a) {
    printf (s"$name <- %d\n",d)
  }
}
The error messages are:

error: value filter is not a member of AIO
  for ((n,d) <- u0.io.a) {
                      ^
error: type mismatch;
 found   : Any
 required: chisel3.core.Bits
    printf(s"$name <- $d\n",d)
                            ^

jackkoenig, 2017-08-07:
You need to iterate on the *elements* of `a`, not `a` itself. Try:

```scala
for ((name, d) <- u0.io.a.elements) { ...
}
```

As for the printf, the type error is because it doesn't know the type of `d` (until you make the fix I mentioned above), but you will get an elaboration time error with it as written. Instead try

```scala
printf(s"$name <- %d\n", d)
```

The above is a C-style printf. Alternatively, you could try using our custom String-interpolator which would be simply:

```scala
printf(p"$name <- $d\n")
```

Note how it uses `p"` instead of `s"`, also that there are no arguments to send after the String. For more information, check out [Printing in Chisel](https://github.com/freechipsproject/chisel3/wiki/Printing-in-Chisel)

apolonic, 2017-08-10:
yes, 
1. "iterate on the elements of the bundle"
2. printf makes RTL $fwrite, for chisel code trap/debug I should use println 
```
class AIO extends Bundles {
  val a0 = Input(Bool())
}
class AM extends Module {
  val io = IO(new Bundle {
    val a = new A().flip
  })
}
class BM extends Module {
  val io = new A()
  val u0 = Module(new AM())
  for ((name,d) <- u0.io.a.elements) {
    println (s"$name  $d\n")
  }
}

```
the result is 

> a0  hisel3.core.Bool@fa

by changing ```s``` to ```p``` the result is 
> Printables(ArrayBuffer(PString(), PString(bready), PString( ), Decimal(chisel3.core.Bool@fa), PString(
>)))

apolonic, 2017-08-10:
@jackkoenig  Could you enlight me how to get other info by this way?
and please forgive me to make it complicated with the string interpolation issue
```
  for ((name,d) <- u0.io.a.elements) {
    case ( getFirrtlDirection($name)) {
      Output: io.a0 := u0.io.a0      // or even io.$name := u0.io.$name
      Input: u0.io.a0:= io.a0
      otherwise: println("Not supported!")
    }
  }
```
If this can be done, then you will also resolve this issue
[https://github.com/freechipsproject/chisel3/issues/661](url)

jackkoenig, 2017-10-19:
Apologies for massive delay in responding to this. Did you ever figure out how to do what you wanted to do?

azidar, 2018-03-16:
Please reopen if issue was not resolved.

closed 2018-03-16 by azidar