chipsalliance/chisel #665: parameterization to the number of IO

apolonic, 2017-08-07:
The following code is what I trend to do, but I got error messages
```
class SRAMIO(np:UInt=3.U) extends Bundle {
  var n = np
  when (n =/= 0.U ) {
    val my_port_$n = Input(Bool())
    n = n - 1.U
  }
}

class sram extends Module {
  val io = IO(new SRAMIO())
}
```

The error message is 
```
- elaborate *** FAILED ***
  chisel3.core.Binding$BindingException: Error: iodef.n: Already bound to OpBinding(sram@0)
  at chisel3.core.Binding$.bind(Binding.scala:100)
  at chisel3.core.Module.Port(Module.scala:84)
  at chisel3.core.Module.IO(Module.scala:113)
  at sram.<init>(top.scala:10)
  ...
- should return the correct result *** FAILED ***
  chisel3.core.Binding$BindingException: Error: iodef.n: Already bound to OpBinding(sram@6)
  at chisel3.core.Binding$.bind(Binding.scala:100)
  at chisel3.core.Module.Port(Module.scala:84)
  at chisel3.core.Module.IO(Module.scala:113)
  at sram.<init>(top.scala:10)
```

apolonic, 2017-08-07:
It seems to me that the var can not be used in Bundle{} or IO(), am I correct?

I also tried this code but the RTL is not as my trend
```
class SRAMIO(my_np:Int=3) extends Bundle {
  for (n <- 0 to my_np ) {
    val my_port_$n = Input(Bool())
  }
  val a = Input(Bool())
}
class sram extends Module {
  val io = IO(new SRAMIO())
}
```
In the RTL there is no my_port_*
```
  sram dut (
    .clock(dut_clock),
    .reset(dut_reset),
    .io_a(dut_io_a)
  ); 
```

apolonic, 2017-08-07:
It seems that declaration in the for/when will be removed. I got the same RTL by this code:
```
class SRAMIO(my_np:Int=3) extends Bundle {
  val a = Input(Bool())
  when (my_np.asUInt > 0.asUInt) {
    val b = Input(Bool())
  }.otherwise {
    val b = Input(Vec(my_np,Bool()))
  }
}
class sram extends Module {
  val io = IO(new SRAMIO(3))
}
```

ducky64, 2017-08-07:
Bundles are meant for static hardware struct definitions, you can use Record if you need dynamic (within certain limits) structs.

Additionally, neither Bundles nor Records can construct hardware. The Binding exception comes because `n` in your code is a hardware type / "wire" (because it's a literal) while IO can only be wrapped around chisel data types (like UInt, etc, that don't have a "value").

Chisel currently is also unable to detect Bundle members that are inside a sub-block, like when / if / for / etc blocks. This has to do with how to detection routine works.

jackkoenig, 2017-08-07:
If you're trying to have a parametrizable number of Bools, have you tried using a Vec?

```scala
class SRAMIO(my_np: Int = 3) extends Bundle {
  val my_port = Input(Vec(my_np, Bool()))
}
```

The names of the ports will be my_port_0 ... my_port_2 or whatever based on my_np

apolonic, 2017-08-10:
Yes, Input(Vec(n, type)) is the best solution. The only limit is the port name has to be the form of string_number

closed 2017-08-10 by apolonic