chipsalliance/chisel #666: parameterization to the name of IO

apolonic, 2017-08-07:
Could you enlight me how to transmit a name to the IO by parameter?

```
class SRAMIO(my_np:Int=3, nm:String="ram1") extends Bundle {
  val a = Input(Bool())
  val b_$nm = Input(Vec(my_np,UInt(2.W)))
}

class sram extends Module {
  val io = IO(new SRAMIO(4,"ram1"))
}
```
how can I tell it the $nm is my parameter?

chick, 2017-08-07:
I think you might want to use the Chisel Record construct instead of Bundle.  There's not a lot of documentation on that (I'll add that to my todo list) but you can see some use in the RecordSpec in the src/test/scala/chiselTests

ducky64, 2017-08-07:
Related: it's theoretically possible to do this in the generated FIRRTL / Verilog, sine Chisel controls the naming there. But the name you see is the name you're going to get in Scala when you need to reference it in Chisel.

As part of the naming system, there was an idea to use string interpolation, for example to name wires generated by a loop. There wasn't any consensus on that, and it hasn't been implemented yet.

jackkoenig, 2017-08-07:
How does the wire looping thing help with this use case? This seems more related to https://github.com/freechipsproject/chisel3/issues/598 and https://github.com/freechipsproject/chisel3/issues/612 (the setName issues).

ducky64, 2017-08-07:
String interpolation in wire names seems like what the original example was trying to do.

jackkoenig, 2017-08-07:
Is it possible to string interpolate variables names in Scala?

ducky64, 2017-08-07:
As far as I know, not in Scala (with the exception of macros), but it would be possible to have Chisel recognize some syntax in generating the names when emitting downstream to FIRRTL.

apolonic, 2017-08-10:
yes , "String interpolation"

azidar, 2018-03-09:
Subsumed by setName or Record.

closed 2018-03-09 by azidar