chipsalliance/chisel #669: Auto clonetype

ducky64, 2017-08-08:
Needing to manually override and implement cloneType is problematic because it's boilerplate (and not necessarily clear how to write one at that) and it doesn't separate implementation and usage (it's an exposed implementation detail, that we're trying to fix with chisel3 but with limited success). So being able to do this automatically would be great.

There are two major approaches:
- `@bundle` macro annotation that automatically generates cloneType. You have to remember to add the `@bundle` annotation, but it doesn't change RTL functionality (so it's closer in spirit to the `@chiselName` annotations). Implementation is more complex (means, it will be a while before this happens), and it's also intertwined with bundle literal constructors.
- Better reflection. The main problem now seems to be that constructor values aren't visible to reflection unless they are declared as `val`s - and a potential workaround is to get users to define their constructor arguments as `val`s in the cloneType error message.

In both cases, there will probably need to be restrictions on what argument types auto-cloneType can work with - namely, being restricted to common immutable types (like `Int`) and Chisel types (like `UInt`, which will be cloned). I imagine this will probably address most places (>90%?) where cloneType is needed, and the user should always be able to override with their own cloneType. Ideally, it would also be possible for the user to specify additional immutable types that cloneType would recognize, though the implementation for that is less clear.

Thoughts?

@sdtwigg @azidar @jackkoenig

jackkoenig, 2017-08-08:
Is there a way with reflection to figure determine which vals were arguments to the constructor? My understanding is that making them vals will add public fields to the class but the actual arguments to the constructor are still private (because they have to be in Java). Perhaps there is some way in Scala reflection?

I am interested in a reflection-based implementation, at a bare minimum it would be better than what we currently have. I still think the macro is the ultimate goal, if only because it would also give us easy and type-safe bundle literal construction.

ducky64, 2017-08-08:
It looks like Scala reflection should be able to get names and types of constructor arguments:
https://stackoverflow.com/questions/27473440/scala-get-constructor-parameters-at-runtime

ducky64, 2017-11-20:
The good news: it appears possible to clone Bundles with val arguments using Scala reflection and matching accessor methods to constructor arguments. 

The bad news: anonymous inner classes are a giant (maybe even unsolvable?) can of worms.

## Anonymous Inner Classes

It turns out that anonymous inner classes (like anonymous Bundles) are going to be problematic. Scala transforms their constructors to explicitly take a reference to the outer class, but I don't see an accessible reference to that using either Java reflection (getDeclaredFields) or Scala reflection (typeSignature.decls).

The way we currently deal with that is by trying to invoke the constructor with all null argument values, which depends on the arguments all not being meaningful. Of course, there aren't actually any checks that any of this is true, that this hasn't been a massive source of NullPointerException errors is kind of impressive. (it's important to note that NullPointerExceptions that occur in the constructor will result in a helpful "you need to write a cloneType" error, but nulls lurking around in value fields that are used later could be the cause of hard-to-debug and very unintuitive errors)

The same null-passing is also not viable for a reliable auto clonetype: if variables in the enclosing are needed (for example, a bitwidth defined in the enclosing Module is needed in the instantiation of a Bundle element), this will NullPointerException out.

Side note: (non-anonymous) inner classes are handled differently: their Scala constructor don't have an extra argument attached to them, though invoking the constructor through Scala reflection requires the outer class reference. It appears that the Java constructor does have the explicit outer class instance argument, though. However, Java reflection (getDeclaredFields) shows a `$outer` method returning the appropriate type, I have not checked if this is accessible and returns a sane value.

Side note 2: In cases where the compiler needs to store a reference to the outer class (for example, a def requires a reference to the enclosing class instance), Scala will generate a `$outer` definition. This probably isn't applicable to most Bundles.

Side note 3: It appears Scala compiler does not generate a "this$0" field (which might have worked in Java), not that that was a particularly elegant idea either.

## Module IO Behavior

It also turns out that IO(...) does not actually clone the input, so it mutably modifies the chisel type passed into it, contrary to all other binding operations (like Wire(...) and Reg(...)). Which may be masking some of the sketchy stuff that the null-happy clone type would otherwise do. The mutable nature of IO(...) could be a source of bugs, though it seems current usage patterns avoid it.

## What next?

Potential options
- Figure out how to access the outer class reference for anonymous inner classes. If anyone knows how to do this, please post it...
- A macros approach to generate cloneType. Requires the user to annotate their bundles with `@bundle` (or something), which means it won't help with legacy code and may see limited use.
- Leave things as-is. Which is probably bad, since nulls may be lurking around and IO(...) behavior is inconsistent with other binding operations.

ducky64, 2017-11-20:
Notes from today's meeting

The changes to IO(...) behavior is the main thing that might break code, and is the main cause of these issues. However, we're lucky in that IO(...) is a method on Module and has access to `this`. The special-case solution (which practically will probably work for most of what people write) is to have IO pass `this` to autoclonetype, which then can check if it typematches the outer class type (and the first Scala constructor argument, which is the convention for anonymous inner classes), and if so, invoke the constructor with that argument.

Potential subtle issues: other uses of cloneType on anonymous inner classes will likely fail, for example if you did `Wire(chiselTypeOf(io))`

Misc additional work remaining: validation of parameter types to restrict autoclonetype parameter filling to those which can be cloned: immutable Scala types (Int, Bool, String, ...) and Chisel types (which should be checked to be unbound, then cloned before being passed into the constructor).

@sdtwigg thoughts about general ways to get the outer class instance given an inner class instance without requiring additional user code?

closed 2018-02-16 by azidar