chipsalliance/chisel #677: Where is implicitConversions needed after #672?

edcote, 2017-08-23:
@jackkoenig Following #672 I hit a compile error in rocket-chip (JtagTap.scala)

    [error] /home/edc/gitrepos/rocket-chip/src/main/scala/jtag/JtagTap.scala:103: type mismatch;
    [error]  found   : chisel3.core.Reset
    [error]  required: chisel3.core.Bool
    [error]   when (reset) {

When compiling chisel, this message came up:

    implicit conversion method resetToBool should be enabled by making the implicit value     scala.language.implicitConversions visible.
    
    This can be achieved by adding the import clause 'import scala.language.implicitConversions' or by setting the compiler option -language:implicitConversions.

I did a few quick hacks adding `-language:implcitConversions` in scalacOptions to both rocket and chisel3 repos, however I recognize that is probably an omission from the previous fix and there might be a quick solutions.

Did you regress rocket-chip master with latest chisel3?

jackkoenig, 2017-08-23:
This is actually an expected error. I'm in the process of making a PR to rocket-chip to work with chisel3 master. rocket-chip has been a little behind but should be caught up soon.

In preparation for future AsyncReset, we changed the type of reset from `Bool` to `Reset`. `Bool` implements `Reset`, but it requires an explicit cast to get a `Bool` from a `Reset`. The solution here is simply `when (reset.toBool) { ...`. 

There is an implicit conversion but that's reserved for `Chisel._` code. I'm not sure why the compiler is suggesting it--that seems unhelpful since the correct answer is to cast.

edcote, 2017-08-23:
Thanks for the background info.  No issues with the .toBool workaround you provided.

closed 2017-08-23 by edcote