chipsalliance/chisel #678: Documentation Needed for RawModule and MultiIOModule

chick, 2017-08-23:
There is no real description in the Wiki on RawModule or MultiIOModule.  We should make
some.  It's possible there should be a section on Experimental, particularly for those features that we would like developers to try out.

Martoni, 2017-08-24:
Has a reference, there is an answered question about [RawModule on stackoverflow](https://stackoverflow.com/questions/41427717/how-to-delete-clock-signal-on-chisel3-top-module). And discussion about this feature [are here](https://github.com/freechipsproject/chisel3/pull/469).

Martoni, 2019-09-10:
I just made a push commit for [documentation proposal](https://github.com/freechipsproject/www.chisel-lang.org/pull/24).

closed 2019-09-11 by seldridge