chipsalliance/chisel #679: a note on the  utility of BlackBox with parameters

apolonic, 2017-08-24:
Hi, some times you need to implement the same blackbox as several instances with different parameters. eg.  the IBUFDS in different IO standards in one project like
```
IBUFDS #(.DIFF_TERM("TRUE"), .IOSTANDARD("ddr3")) ibufds_ddr3 (
  .IB(ibufds_IB_ddr3),
  .I(ibufds_I_ddr3),
  .O(ibufds_O_ddr3)
);
IBUFDS #(.DIFF_TERM("TRUE"), .IOSTANDARD("lvds")) ibufds_lvds (
  .IB(ibufds_IB_lvds),
  .I(ibufds_I_lvds),
  .O(ibufds_O_lvds)
);
```
I may define a IBUFDS BlackBox like
```
import chisel3._
import chisel3.experimental._ // To enable experimental features

class IBUFDS extends BlackBox(Map("DIFF_TERM" -> "TRUE",
                                  "IOSTANDARD" -> "DEFAULT")) {
  val io = IO(new Bundle {
    val O = Output(Clock())
    val I = Input(Clock())
    val IB = Input(Clock())
  })
}
```
Then I don't know how to use the `val params: Map[String, Param]` as a parameter. (I got a workaround but any better solution?)

I tied to change the define as this
```
import chisel3._
import chisel3.experimental._ // To enable experimental features

class IBUFDS(val myparams: Map[String, Param] = Map.empty[String, Param]) extends BlackBox(myparam)) {
  val io = IO(new Bundle {
    val O = Output(Clock())
    val I = Input(Clock())
    val IB = Input(Clock())
  })
}
```
then instance it as
```
import chisel3._
import chisel3.experimental._ // To enable experimental features

val IBUFDS_ddr3 = Module(new IBUFDS(Map(
   "DIFF_TERM"->fromStringToStringParam("TRUE")
   ,"IOSTANDARD"->fromStringToStringParam("ddr3")
)))
val IBUFDS_lvds = Module(new IBUFDS(Map(
   "DIFF_TERM"->fromStringToStringParam("TRUE")
   ,"IOSTANDARD"->fromStringToStringParam("lvds")
)))
```

azidar, 2018-03-09:
Solution: add more documentation about this, and potentially an example in the generator bootcamp

chick, 2018-12-17:
Done, in wiki

closed 2018-12-17 by chick