chipsalliance/chisel #685: result width of multiplication

apolonic, 2017-08-30:
chisel code:
```
val x = Wire(UInt(64.W))
val z = Reg(UInt(128.W))
z := x * x
```
verilog code:
```
reg [127:0] z;
wire [63:0] x;
always...
  z = x * x;
...
```
the risk is the carrier could be lost according to Table 5-22—Bit lengths resulting from self-determined expressions, verilog IEEE std 1394-2005, in which it is specified that 

> i op j, where op is:+ - * / % & | ^ ^~ ~^ , the result bit lengh is max(L(i),L(j))

As so far, the verilog code can simulated correctly on Synopsys tool (the synthesis result might be also correct, to be checked), but failed on Xilinx tool ! 

Workaround:
extend the operator's width, let the verilog interpreter to optimize the logic.
```
val x = Wire(UInt(64.W))
val xl = Wire(UInt(128.W))
xl := x
val z = Reg(UInt(128.W))
z := xl * xl
```

aswaterman, 2017-08-30:
I'm pretty sure the Verilog is correct and the Xilinx tool is wrong

closed 2018-03-16 by azidar

udif, 2024-08-14:
I know this case is closed for 6 years, but for the benefit of those getting here, the reason this is a Xilinx bug is due to section 5.5.2 in IEEE2364-2005.
The width of `a * b` is `Max(len(a), len(b))` as this is what the standard calls "self determined expressions" (Table 5-22 in the standard).
However, in the case above, `a = b * c;` is what the standard calls a "context-determined expression", and the width is dictated (in this case) by the width of the LHS.
The same logic applies to the current IEEE1800-2023.