chipsalliance/chisel #686: Contributing to the Wiki docu

schoeberl, 2017-08-30:
I would like to contribute some words to the Wiki. Currently I have some items on the Chisel 2 to 3 migration path. However, I could not find out how to do a pull request towards the Wiki files. And I don't have write access.

I could clone the Wiki git, generate patches, and send them per email to someone with write access to the project. However, this flow is a little bit heavy weight for both sides.

Cheers,
Martin

ccelio, 2017-08-30:
I think you can do pull requests against this repo:

https://github.com/ucb-bar/chisel3-wiki

schoeberl, 2017-08-30:
Ok, I see. So the chisel 3 repo's Wiki is actually based on (backed up by?) a GitHub repo in ucb-bar? Interesting.

Will try in a moment. Thanks.

schoeberl, 2017-08-30:
worked out as suggested by @ccelio.

closed 2017-08-30 by schoeberl