chipsalliance/chisel #687: simulation issue of "assert":  Error-[SVA-SSTINSC] SVA-system task in non-SVA context.

apolonic, 2017-08-31:
chisel scala code:
assert(aaa,"bbb")
RTL code:
```
    `ifndef SYNTHESIS
    `ifdef PRINTF_COND
      if (`PRINTF_COND) begin
    `endif
        if (aaa) begin
          $fwrite(32'h80000002,"Assertion failed: bbb\n    at aaa,\"bbb\")\n");
        end
    `ifdef PRINTF_COND
      end
    `endif
    `endif
    `ifndef SYNTHESIS
    `ifdef STOP_COND
      if (`STOP_COND) begin
    `endif
        if (aaa) begin
          $fatal;                // Synopsys VCS reports error here
        end
    `ifdef STOP_COND
      end
    `endif
    `endif
```
elaboration error log:
```
Error-[SVA-SSTINSC] SVA-system task in non-SVA context.
  Attempt to use a SVA-system task in non-SVA context.
  "....", line#: $fatal;
```
Is this a newer version issue of systemverilog assertion language?

joisty, 2017-11-20:
I am facing exactly the same problem.

azidar, 2018-03-16:
Please use the system verilog flag for vcs.

closed 2018-03-16 by azidar