chipsalliance/chisel #697: Comparison Operators should check whether widths allow both outcomes.

chick, 2017-09-25:
It would be nice if comparison operators could check information available to there operands that would determine if both outcomes of the operator are possible.
A  common early programming error in chisel is.
```scala
val counter = RegInit(0.U)
when(counter > 15.U) {
  ???
}
```
```counter``` above cannot get bigger than 1 since it's width was not set explicitly and thus the when block will never be executed.  I don't know if this might be a new feature of *constant propagation* or maybe *dead code elimination* but I think this would help cut off a number of easily made errors.

ccelio, 2017-09-30:
Chisel, at its heart, is a generator language. We (and every other language) already remove dead code that a user may have intended to hook up, and I see this as the same situation.

With that in mind, it would be nice to provide the user a way to check for coding errors. Things like `AssertStillLive(io.req.ready)` (i.e., things that check that a wire isn't pruned or left floating or checks that a condition is possible).

chick, 2019-06-29:
@jackkoenig What do you think about converting this to an RFC?

azidar, 2019-12-02:
Not an issue - decided that we can't catch all of these by solely looking at the expression.

closed 2019-12-02 by azidar