chipsalliance/chisel #699: printf issue

codelec, 2017-10-01:
printf's are not generating the appropriate fwrite in the corresponding the verilog 
for eg. 
https://github.com/librecores/riscv-sodor/blob/master/src/rv32_1stage/dpath.scala#L194 
will generate 
          $fwrite(32'h80000002,"Cyc???d?Op????x?h??Op????x?h??W??c??d???x?h???c?Mem??d??R??x?h?W??x?h??PC???x?h??c?c?DASM??h?\n",_T_321,alu_op1,alu_op2,_T_324,wb_addr,wb_data,_T_327,io_ctl_wb_sel,io_dmem_resp_bits_data,io_dmem_req_bits_data,pc_reg,_T_330,_T_351,io_imem_resp_bits_data);

jackkoenig, 2017-10-01:
My apologies for this, I introduced this bug in https://github.com/freechipsproject/firrtl/pull/663 and fixed it in https://github.com/freechipsproject/firrtl/pull/668.

codelec, 2017-10-02:
thanks .. when would these changes make it into maven since the current package from maven doesn't seem to have these changes

closed 2017-10-02 by codelec

ucbjrl, 2017-10-02:
We should be pushing out another SNAPSHOT release by 10/6/2017.

jackkoenig, 2017-10-02:
We are also planning to push out a release candidate for version 3.0.0 this week!

In the immediate term you can use the previous SNAPSHOT release from 9-14: https://oss.sonatype.org/#nexus-search;gav~edu.berkeley.cs~chisel3_2.11~~~~kw,versionexpand