chipsalliance/chisel #701: Unprotect def getModulePorts

hcook, 2017-10-03:
As we go down the path of `Record` and programmatic bundle creation, I more often want to get access to all the ports of a `RawModule` in scala-land. I know it is perhaps scary because the `Module` needs to be closed first, but maybe there could be a built-in exception if that is not the case?

I specifically want to be able to have implicit classes extending modules with e.g. `dontTouchPorts` methods, or other such annotations :)

ducky64, 2017-10-03:
I think there was the idea of bulk connecting to a Module meaning bulk-connecting to all its ports. I don't know if we wanted to go somewhere with that, and what the behavior would be with implicit ports like clock and reset.

Can you give a (few?) example use cases and maybe proposed syntax? I'm not sure exposing everything is the best solution, maybe there's some good limited API that would meet the use case?

jackkoenig, 2017-10-03:
A couple of use cases
1. Wrap a RawModule and pass all of it's IO through. Need the ability to gather all of the ports, clone their type, and bulk connect them.
2. Annotate all ports of a module as @hcook mentioned. Marking the ports dontTouch is obviously very important for exposed interfaces and there are other things you might want to do as well.

We already have an existent API for the ports of a regular Module [`chisel3.getModulePorts`](https://github.com/freechipsproject/chisel3/blob/master/src/main/scala/chisel3/package.scala#L343), there really ought to be some way for RawModules as well. We could make the type of the "better" API just to return an anonymous Record type that includes all of the ports of the Module. This Record could then be used for bulk connection and cloned types.

jackkoenig, 2017-10-10:
I have been informed that there is actually a public API here. RawModule aliases chisel3.core.UserModule which has a field [getPorts](https://chisel.eecs.berkeley.edu/api/index.html#chisel3.core.UserModule@getPorts:Seq[chisel3.internal.firrtl.Port]).

I think this API leaves much to be desired but it works.

jackkoenig, 2017-10-10:
Anyway, while the API could be better, I think this particular issue is resolved.

closed 2017-10-10 by jackkoenig