chipsalliance/chisel #703: Extend API to include explicit handling of reset and initialization

chick, 2017-10-10:
The Invalidate API has exposed problems with apparently uninitialized wires, particularly when using DeqIO and EnqIO's initialization. 
Proposed that we create an ```OnReset``` operator that essentially implements ```when(reset.toBool)```, but that might also implement the block without the surrounding when which could take advantage of *last connect semantics* to override these defaults

jackkoenig, 2017-10-10:
Can you provide a more specific example to help better illustrate the problem? I don't really understand what the issue is.

The proposed `OnReset` sounds to me like `when(Module.reset)` ([ScalaDoc](https://chisel.eecs.berkeley.edu/api/index.html#chisel3.core.Module$@reset:chisel3.core.Reset)). I don't see how "last connect semantics" relate here without returning a reference to the reset used so that it can be connected to, but `Module.reset` does properly use the current implicit reset (which can be set beforehand via `withReset` and `withClockAndReset`)

chick, 2017-10-10:
The example is Router which used to have a ``when(reset.toBool)``` at the beginning that called 
```noenq()``` on each of it's outputs.  When the invalid API was turned on an error was generated because there was an uninitialized Wire. It ends up being something like.
```
io.out(n).bits := Mux(io.in.valid, io.in.bits, Mux(reset, 0, INVALID))
```
In practice, the you'll never get to the INVALID, but firrtl doesn't know that.

You could fix this by actually doing the ```noenq``` in addition to the reset early in the module.
This makes the INVALID go away for firrtl, but last connect semantics makes you get the behavior you want for the IO.

The point being that DeqIO and EngIO have these weird conflicts hidden them, which is probably why nobody uses them, and that's bad because there ought to be clear tools for implementing decoupled IO.  Init and Reset behavior is one aspect of what they need to be rationally used but that has proven difficult.  Reset is important to a lot of circuits but there are no supporting tools for using it in a way that gets things right.  Jim and I fell victim to this when we just removed the ```when(reset.toBool)``` and got the initialization right but screwed up the ability to reset. I think we both feel that having a simple model for acknowledging the desired reset behavior of the module would avoid errors and remind people to test reset behavior.  They do this, ad hoc, in big projects but we don't have much to offer people getting started in Chisel

ucbjrl, 2017-10-10:
It looks like for sake of simplicity and cleanliness, you want something like:
```scala
class MyModule extends Module {
  val io = IO(new Bundle {
    val in = Input(Bool())
    val out = Output(Bool())
  })

  def onReset(): Unit = {
    io.out := false.B
  }

  onReset()
  when(reset.toBool) {
    onReset()
  }
  ...
```
That is, you need to be able to evaluate the onReset code outside of a conditional (so firrtl doesn't complain about "io is not fully initialized") and you also want to evaluate the same code when reset is asserted.

closed 2018-03-02 by azidar