chipsalliance/chisel #704: Erroneous "Module without wrapping it in Module()"-Exception?

jkorinth, 2017-10-11:
The code below raises an
`chisel3.internal.ChiselException: Error: attempted to instantiate a Module without wrapping it in Module()`
error, even though the module _is_ wrapped in a module. How to fix this?

```
import chisel3._

class Test extends Module {
  val io = IO(new Bundle)
}

object Test {
  def main(args: Array[String]) {
    val f = () => new Test
    val x = Module(f())     // <---- why does this raise an Exception?
  }
}
```
This does not work either:
```
val f = () => Module(new Test)
val x = f()
```

jackkoenig, 2017-10-11:
The error message is actually just wrong. The issue here is that you are attempting to instantiate Chisel modules outside of the standard Chisel invocation. There is global state we have to set up and manage so you can't just instantiate Chisel Module classes. You should use the [Driver](https://chisel.eecs.berkeley.edu/api/index.html#chisel3.Driver$) like so:

`chisel3.Driver.execute(args, () => new Test)`

None of proper ways of using the Driver to invoke Chisel return the Module object (for various reasons), is there a particular reason why you want the object?

edwardcwang, 2017-10-11:
Maybe we want to track this in a better way for more user-friendly error messages? @ucbjrl @ducky64 

From an internals engineering perspective, the builder creates a new context (which is how Chisel keeps track of the `readyForModuleConstr` flag).

https://github.com/freechipsproject/chisel3/blob/802cfc4405c28ae212a955a92c7a6ad2d2b6f0c2/chiselFrontend/src/main/scala/chisel3/internal/Builder.scala#L238

If you just create a bare Module with no context whatsoever, then the dynamicContext call fails and just keeps creating a new one, which means that the flag never gets stored.

https://github.com/freechipsproject/chisel3/blob/802cfc4405c28ae212a955a92c7a6ad2d2b6f0c2/chiselFrontend/src/main/scala/chisel3/internal/Builder.scala#L165

jackkoenig, 2017-10-12:
Yeah I think this actually used to have a better error message so we should alert users if they aren't invoking Chisel properly.

I believe that @chick added the getOrElse for the dynamicContextVar in order to do Chisel-y things in the REPL. I believe it is pretty dangerous since it leads to thinks where Chisel kind of runs (but incorrectly) when instead it should just alert the user that it can't handle this. I think we should change that to getOrElse throw a useful exception.

jkorinth, 2017-10-12:
@jackkoenig Ah, I see! I'm using a [packaging script](https://github.com/jkorinth/chisel-packaging) to generate an [IP-XACT](http://www.accellera.org/downloads/standards/ip-xact/) description for the generated Module automatically. Some relevant properties, e.g.,

  * presence/absence of module ports
  * presence/absence of configuration registers
  * bit widths of ports, ...

are configuration-dependent and I need some way to query the configuration of the generated module (i.e., the constructor arguments). In Chisel 2.x I could get the Module instance, so I thought it would be easiest to implement query methods on the object itself. I changed the approach to use a configuration object instead, which I think is cleaner anyway, but was curious about the error message.

Thanks for the clarification! :thumbsup:

jackkoenig, 2017-10-19:
@jkorinth Thank you for the information about your use case! Using the constructed object itself to query details is certainly a popular use case that we should support. Thank you for the report!

closed 2017-10-19 by jackkoenig

makslevental, 2021-05-26:
the [docs](https://www.chisel-lang.org/chisel3/docs/resources/faqs.html#get-me-verilog) show this

[image]

where i have

```scale
object Im2Col extends App {
  val qSize = 15
  val stride = 1
  val padding = true
  val fifo = true
  val dWidth = 16
  val throughput = 1
  val inputCycle = 1
  val imgSize = 32
  val inChannels = 1
  val kernelWidth = 3

  val im2col = new Im2Col(
    UInt(dWidth.W),
    math.pow(2, imgSize).toInt,
    inChannels,
    2 * kernelWidth + 1,
    qSize,
    stride,
    padding,
    throughput,
    inputCycle,
    fifo = fifo
  )

  (new chisel3.stage.ChiselStage).emitVerilog(im2col)
}
```

and i hit this same error message

```
Exception in thread "main" java.lang.IllegalArgumentException: requirement failed: must be inside Builder context
	at scala.Predef$.require(Predef.scala:281)
	at chisel3.internal.Builder$.dynamicContext(Builder.scala:339)
	at chisel3.internal.Builder$.readyForModuleConstr(Builder.scala:499)
	at chisel3.experimental.BaseModule.<init>(Module.scala:168)
	at chisel3.RawModule.<init>(RawModule.scala:20)
	at chisel3.MultiIOModule.<init>(RawModule.scala:163)
	at chisel3.internal.LegacyModule.<init>(RawModule.scala:198)
	at systolic.Im2Col.<init>(Im2Col.scala:23)
	at systolic.Im2Col$.$anonfun$im2col$1(Im2Col.scala:178)
	at chisel3.internal.plugin.package$.autoNameRecursively(package.scala:52)
	at systolic.Im2Col$.delayedEndpoint$systolic$Im2Col$1(Im2Col.scala:168)
	at systolic.Im2Col$delayedInit$body.apply(Im2Col.scala:150)
	at scala.Function0.apply$mcV$sp(Function0.scala:39)
	at scala.Function0.apply$mcV$sp$(Function0.scala:39)
	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:17)
	at scala.App.$anonfun$main$1$adapted(App.scala:80)
	at scala.collection.immutable.List.foreach(List.scala:431)
	at scala.App.main(App.scala:80)
	at scala.App.main$(App.scala:78)
	at systolic.Im2Col$.main(Im2Col.scala:150)
	at systolic.Im2Col.main(Im2Col.scala)

Process finished with exit code 1
```